raspberrypi/noobs #317: Samsung Pro Micro SD Card 64GB Noobs Wont boot

kryptalivian, 2016-02-26:
I'm using the card stated above and noobs will not boot. Amber and Green light stay solid. :-( Tried both Noobs and Noobs Lite

lurch, 2016-02-26:
Which model of RaspberryPi?  Do you have the exact MicroSD card model-number?
I assume this is with the latest NOOBS v1.7.x ?  Does the same card work okay with 'standalone' [Raspbian](https://www.raspberrypi.org/downloads/raspbian/) ?

kryptalivian, 2016-02-26:
Noobs 1.7.0 
Raspberry Pi 2
model of SD card mb mg64d

kryptalivian, 2016-02-26:
Not sure on raspbian. I usually use sd cards for Gaming distros like lakka and recalbox

NicoHood, 2016-02-26:
I think the pi does not support >32gb sd cards at all.

kryptalivian, 2016-02-26:
I'm using 128gb currently on my pi :-/

lurch, 2016-02-26:
> I think the pi does not support >32gb sd cards at all.

It does. See https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md for more explanation ;-)

lurch, 2016-02-28:
The closest I currently have to your problematic card is a Samsung Pro 32GB microSD card (model number MB-MG32D) and I've just checked and it works absolutely fine with NOOBS Lite v1.7 on a Raspberry Pi 2.

Other questions:
Is it only this card that has problems with NOOBS v1.7 (i.e. are you able to boot NOOBS 1.7 okay from other microSD cards on the identical hardware) ?
Do you get *any* display output at all (Rainbow square, error messages, etc.) or does the display stay blank?
Are you using a "quality" power-supply? (unlikely to be the problem, but it doesn't hurt to check)

If you can try using standalone Raspbian on the same microSD card, and confirm / deny whether that works too, that'd be very useful. Thanks!

Etuldan, 2016-10-08:
Same issue, same SD card.
Nothing at all on boot, no led on keyboard, no HDMI display (black screen) (but my screen detect that something has been plugged).

FAT32 formatting.

Was OK with another distribution (this one : https://docs.cozy.io/en/host/install/install-on-raspberry.html ) with direct img installation (no Noobs)

The NOOBS I got is from RecalBox.

Etuldan, 2016-10-09:
Ok.
May be not a noobs issue, as:

Was trying with Windows10, even with "MiniTool Partition Wizard Free Edition", checked that it was FAT32, but KO.
I then try with gparted (inside a LMDE distribution on dual boot), FAT32, and it's working.

ghollingworth, 2016-10-09:
The reason it won't boot is likely to be because Windows is partitioning the disk using a GUID partition scheme and also actually making it exFAT (they absolutely refuse to program FAT...  Wonder if the reason why is something to do with the license they charge to use exFAT?)

One or both of these things!  But actually it should work even with GPT as long as it's FAT16/32

Gordon

retrostudd, 2016-10-24:
I was having a similar issue with my 64GB Kingston SD.  Formatted with GParted Fat32 fixed me right up.

closed 2017-01-19 by XECDesign

JayBk, 2018-01-11:
Since this is one of the very first results on Google when searching something related to 64GB (or anything 32GB+) SD cards on Raspberry Pi's, I thought I would add the following definitive information that is currently posted by the Raspberry Pi Foundation: https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md (the link @lurch shared)..

They say: "According to the SD specifications, any SD card larger than 32GB is an SDXC card and has to be formatted with the exFAT filesystem. This means the official SD Formatter tool will always format cards that are 64GB or larger as exFAT.

The Raspberry Pi's bootloader, built into the GPU and non-updateable, only has support for reading from FAT filesystems (both FAT16 and FAT32), and is unable to boot from an exFAT filesystem. So if you want to use NOOBS on a card that is 64GB or larger, you need to reformat it as FAT32 first before copying the NOOBS files to it."