raspberrypi/noobs #318: 1.8.0 Jessie or Wheezy?

tuupola, 2016-03-02:
Does the latest 1.8.0 version include Raspbian Jessie or Wheezy? If Jessie then what is the last version which had Wheezy? [Releases page](https://github.com/raspberrypi/noobs/releases) is bit out of date.

procount, 2016-03-02:
According to the Raspbian releases, Jessie was available on 25-09-2015, so all versions of full NOOBS contained Jessie since then, I think Jessie started with v1.4.2. So any version of NOOBS full prior to that contained Wheezy.

But I agree with you about the documentation - please could the source & documentation for NOOBS and release_notes be updated more promptly? NOOBS readme.md  has not been updated to take account of any of the new features since v1.5.0
Noobs v1.6 and v1.7 contained no release_notes.txt for Raspbian, so it is difficult to keep track of the history. But I'm glad to see them back in v1.8.0.

lurch, 2016-03-02:
AFAIK, the Raspbian version within NOOBS is always kept in-step with the standalone version of Raspbian.

To download the older Wheezy releases, see the comments here: https://www.raspberrypi.org/blog/four-years-of-pi/#comment-1253609

tuupola, 2016-03-02:
Images can also be found from here: http://downloads.raspberrypi.org/NOOBS/images/

NicoHood, 2016-03-02:
And where is the source for NOOBS 1.8?

closed 2016-03-26 by tuupola