raspberrypi/noobs #320: Recovery Mode Keyboard USB Placement

MonacleforSauron, 2016-03-06:
I recently tried to put my pi into Recovery Mode, but nothing happened when I pressed shift, it just started up like normal. I tried over and over, and I tried pressing the shift key a lot, pressing both shift keys, unplugging the USB and putting it back in, etc. Finally, I saw somewhere that I should put the keyboard in the the tob USB (I had it in one of the bottom two), and it worked. I'm not sure why this happened. Do the USBs start up at different times?

lurch, 2016-03-07:
Blimey, I've never heard of that before.

What model of Pi are you using? What exact make & model of keyboard are you using? Is it wired or wireless? Does the keyboard work on all four USB ports once Raspbian has loaded? Once you've triggered and got past the 'press shift' stage within NOOBS (i.e. you're looking at the OS-install-list menu) can you then swap the keyboard into each of the USB ports in turn, and do the up/down arrow keys still work with each port?

Do you have a different USB keyboard that you can repeat the above experiments with, for the sake of comparison?

XECDesign, 2017-11-03:
No response

closed 2017-11-03 by XECDesign