raspberrypi/noobs #321: Support for Raspberry Pi 3 WiFi?

matthuisman, 2016-03-07:
I looked through and surprised I couldn't find this as an issue??

I know the official WiFi dongle works in latest NOOBs, but it doesn't seem to detect the Raspberry Pi 3 built-in WiFi.

I assume this is planned?

voltagex, 2016-03-08:
Wifi was greyed out for me. Running through the installer now - haven't tried the "network only" installer but I'd bet it'd be the same.

voltagex, 2016-03-08:
Questions to be answered:

* What impact would bumping the NOOBS kernel to 4.1.x have? I'm not sure what exact revision is required for the BCM43438 chip to function but http://raspberrypi.stackexchange.com/questions/43473/does-the-bcm43438-wifi-chip-in-raspberry-pi-3-support-power-management-setting suggests we might need something even newer than that

* What's required to get NOOBS to distribute the proprietary firmware blob that the chip needs?

maxnet, 2016-03-08:
Just needs bcm2710-rpi-3-b.dtb added to the FAT partition.

matthuisman, 2016-03-08:
@maxnet SPOT ON!
That did it.

So easy.
Just pulled the bcm2710-rpi-3-b.dtb from latest Raspbian image and put into root folder of NOOBs.
Boots up and connects to WiFi!

I've uploaded the file here for anyone who can't be bothered finding it :)
https://mega.nz/#!aM5kiJCJ!uNa66VwgnTL6yhgqSGYD2gi84BBoBZmXzabL8jwmVHc

Wonder why they didn't include that file? 
Maybe just missed?

voltagex, 2016-03-09:
@maxnet if you can tell me what controls the kernel+dtb that's created from BUILDME.sh I can make a pull request

lurch, 2016-03-09:
@voltagex I guess you want to look at https://github.com/raspberrypi/noobs/blob/master/buildroot/package/rpi-firmware/rpi-firmware.mk

matthuisman, 2016-03-11:
I made a pull request for the fix

https://github.com/raspberrypi/noobs/pull/323

lurch, 2016-03-21:
Fixed by https://github.com/raspberrypi/noobs/commit/b7d838d5b0cc2db436c5ca1265d9999bb5b53b85 and released in NOOBS v1.9

closed 2016-03-21 by lurch

HackG, 2016-04-20:
does anyone know why I wouldnt be able to add this **bcm2710-rpi-3-b.dtb** file to my NOOBs sd card? It seems write protected and I tried the write protection switch on and off (trying to read/write from macbook pro 2012, internal card reader)

btw I uploaded it to share here as well - https://drive.google.com/open?id=0B_2KE9k39l-yWE1KS29JNk10blk

lurch, 2016-04-21:
Try using the latest version of NOOBS (v1.9 or above) - it already includes the files needed for WiFi on the Pi 3, so no extra files need to be added.