raspberrypi/noobs #324: NOOBS 1.8 source?

procount, 2016-03-11:
@maxnet - Do you have any idea when you will be able to release the source for v1.8?
It is nearly 2 weeks since the Pi3 launch and I am waiting to update my fork.

maxnet, 2016-03-11:
Don't think anything has changed other than the bumped version number, but would have to ask.
(binary releases are not done by me)

procount, 2016-03-11:
(Sorry i was not aware of whose responsibility it was)
A bumped major version number indicates a change to the NOOBS source, otherwise it should have been v1.7.1 to indicate just a new OS.
At the very least, pull request #323 should be included to enable the onboard wifi. I don't know if this also required the wifi driver, or if it was already in v1.7. (I thought that v1.7 just included the wifi driver for the external RPF dongle as it was released before the Pi3 and there were comments on the forum about the wifi driver not being available in time...?)
The boot files & kernel files seem to be slightly different too.
But in any case, it would be great if the github source to be kept up to date with the release so we can see the changes for ourselves. Currently my mods are on v1.7 and I don't want to release based on that only to find v1.8 sources get released the day later and I have to rebase all my mods again.
And it would be really nice if the README.md file and release information could be updated to indicate the changes since v1.5. i know most of them already, but newcomers are still looking at outdated information and will not be aware of the new features.
(Maybe I should write my interpretation and submit a pull request if I have time)

maxnet, 2016-03-11:
>A bumped major version number indicates a change to the NOOBS source, otherwise it should have 
>been v1.7.1 to indicate just a new OS.

Yes, there has been a minor change that hides operating systems not supported from the list.
But that's already checked into this repository.

Like I said, I think the only thing missing is the version number bump itself that is typically done before release.

lurch, 2016-03-11:
> The boot files & kernel files seem to be slightly different too

That would also be enough to warrant a 'major version' bump, as it would necessitate a new version of NOOBS Lite.

procount, 2016-03-11:
I checked and the pi3b wifi driver has been included in v1.8.
So I would still appreciate it if this github is updated so I can build an exact image of v1.8 instead of trying to guess any other build changes.

maxnet, 2016-03-11:
>I checked and the pi3b wifi driver has been included in v1.8.

What makes you think that is new?

procount, 2016-03-11:
It is newly included in v1.8 of NOOBS, When I tried v1.7 on my Pi3B with the 2710-RPI3.dtb applied, the wifi did not work, but applying the same DTB to v1.8, the wifi now works. So I conclude the driver has been added.  I would like to add it to my version of the kernel. Which wifi driver is it? Are there any other drivers or kernel changes added in v1.8? WJDK. 

This github contains not just the source to recovery (which seemingly has not changed) but the configuration of the kernel, drivers and other packages. So it would be really nice if this github could be updated on every major release so that we can build an exact image of NOOBS.
I'm sure this must be in a local git repo, so isn't it just a question of pushing it up to github? Shouldn't that be done as part of the binary release? (or at least soon afterwards to avoid any spoilers in the case of new models)

maxnet, 2016-03-11:
>It is newly included in v1.8 of NOOBS, When I tried v1.7 on my Pi3B with the 2710-RPI3.dtb applied, the 
>wifi did not work but applying the same DTB to v1.8, the wifi now works. So I conclude the driver has been added

I think you shouldn't jump to conclusion so fast.
Did you try an official v1.7 NOOBS version, or your own custom build?
Because it seems like you disabled it yourself in your own tree...

https://github.com/procount/noobs/commit/ebc50521aa162253e667a709dc6f92f85030831c#diff-1c8f157a08c2f2c03833567482a7c354L65

procount, 2016-03-11:
I'm pretty sure it was an official v1.7.
The version you linked to on my custom build ends in _150, which means it is based on v1.5. (Jan 29)
My github is a bit out of date. I have updated locally to v1.7, and was just about to push it when v1.8 came out. So I'm just waiting to rebase on the new source.

procount, 2016-03-11:
I can try it again later just to make sure.
But if you are saying it was already included in v1.7, then what were the changes at binary release that required a bump of version number to v1.8? I can't be sure I'm not missing something without an updated repo.

procount, 2016-03-11:
Very strange & interesting. Thanks for spotting that. I did indeed accidentally delete it as long ago as v1.5. But I wasn't expecting that change to have been done so long ago - hence my conclusion.

But I'm still confused as to why the version was bumped if there were no changes?
Am I safe to release my code code based on v1,7 if I include the rpi3 DTB file?

maxnet, 2016-03-11:
>But I'm still confused as to why the version was bumped if there were no changes?

Again, the change is that operating systems not supported by the Pi model the user has are now hidden.
Commit that does that is already here, so there are no changes to what is on github...
But it was not part of 1.7, so there does is a change to 1.7...

procount, 2016-03-11:
OK Thanks for your explanations.  I understand now.
I think it would have avoided confusion though if v1.8 had been released. Then we would know that there were few to no changes and wouldn't be waiting for the "source code to be tidied up" again before it was released, which I had presumed was the case again (wrongly).

Thanks for being patient with me -  I was used to the github being updated for each version, so this was a bit unusual.
I'll get back to updating my mods now :-)

Ruffio, 2017-01-13:
@procount is this still an issue? If resolved, please it. Thanks.

closed 2017-01-19 by XECDesign