raspberrypi/noobs #331: INSTRUCTIONS-README.txt should mention 64GB Ex-Fat on mac

chacham2, 2016-03-22:
Noobs OSX INSTRUCTIONS-README.txt says to download the SD Card formatter and let it run. It makes a 64GB SD Card Ex-FAT, and the pi won't boot. Disk Utility can change it to FAT32, which works beautifully.

INSTRUCTIONS-README.txt ought to mention this.

lurch, 2016-03-22:
Would adding a link to https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md be enough, or would you like the information duplicated in NSTRUCTIONS-README.txt

chacham2, 2016-03-22:
I would suggest duplication as the user might not have an Internet
connection, or perhaps see the importance of the link without a decent
explanation of the issue. In which case, it'd just be easier to repeat the
instructions already.

XECDesign, 2017-01-19:
I haven't duplicated the instructions, but have added a link. Given that it's an edge case, I don't want to add yet another place the instructions would need to be maintained.

closed 2017-01-19 by XECDesign