raspberrypi/noobs #332: [Question] Languages

ppigmans, 2016-03-24:
Hi there! lately i worked on a dutch translation for RC_gui, it was accepted to my amazement, now i wish to translate NOOBS aswell, but i have no idea where to start, can anyone point me in the right direction?

Thanks in advance

lurch, 2016-03-24:
https://github.com/raspberrypi/noobs/blob/master/README.md#addingupdating-translations

maxnet, 2016-03-24:
Shouldn't have to add Dutch as new language as it was the first translation ever added to NOOBS...
Can update the lines marked 'unfinished' though

https://github.com/raspberrypi/noobs/blob/master/recovery/translation_nl.ts

ppigmans, 2016-03-24:
BLEEEEEERGH this pc im working on is slow as heck, ill work on it tonight, see what i can finish and then do the rest in the weekend :) it will be finished Before sunday evening i hope

ppigmans, 2016-03-25:
the translation is done :)

closed 2016-03-25 by ppigmans