raspberrypi/noobs #335: Segmentation fault SIGSEGV libwebkitgtk-3.0.so.0

alexanderpublit, 2016-04-04:
When a dashboard slideshow on Klipfolio is started via Epiphany, included in NOOBS, it crashes:

```
** Message: console message: https://app.klipfolio.com/a/js/datasource.core?rv=456181910 @155: [17:47:54]: connection state change:

** Message: console message: https://app.klipfolio.com/a/js/datasource.core?rv=456181910 @156: [17:47:54]: Connected (waiting for provisioning)

[Thread 0x5af1d280 (LWP 2800) exited]
** Message: console message: https://app.klipfolio.com/a/js/datasource.core?rv=456181910 @155: [17:47:55]: connection state change:

** Message: console message: https://app.klipfolio.com/a/js/datasource.core?rv=456181910 @162: [17:47:55]: Provisioned

[Thread 0x5d71d280 (LWP 2804) exited]
[Thread 0x5b71d280 (LWP 2801) exited]
[Thread 0x5c71d280 (LWP 2799) exited]
[Thread 0x5cf1d280 (LWP 2779) exited]
[Thread 0x5f8b2280 (LWP 2802) exited]
[Thread 0x5f0b2280 (LWP 2798) exited]
[New Thread 0x5f0b2280 (LWP 2805)]
[New Thread 0x5f8b2280 (LWP 2806)]
[New Thread 0x5cf1d280 (LWP 2807)]
[Thread 0x600b2280 (LWP 2803) exited]
[Thread 0x5f0b2280 (LWP 2805) exited]
[Thread 0x5f8b2280 (LWP 2806) exited]

(epiphany-browser:1163): GLib-CRITICAL **: Source ID 1404974 was not found when attempting to remove it
[Thread 0x5cf1d280 (LWP 2807) exited]
[New Thread 0x5cf1d280 (LWP 2809)]
[Thread 0x5cf1d280 (LWP 2809) exited]
[New Thread 0x5cf1d280 (LWP 2811)]
[New Thread 0x5f8b2280 (LWP 2812)]
[Thread 0x5cf1d280 (LWP 2811) exited]

Program received signal SIGSEGV, Segmentation fault.
0x7555f4f8 in ?? () from /usr/lib/arm-linux-gnueabihf/libwebkitgtk-3.0.so.0
(gdb) bt
#0  0x7555f4f8 in ?? () from /usr/lib/arm-linux-gnueabihf/libwebkitgtk-3.0.so.0
#1  0xe7a52108 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

Epiphany-browser: v.3.8.2
Powered by WebKit 2.4.1

lurch, 2016-04-04:
I think you actually mean Raspbian (which gets installed by NOOBS) rather than NOOBS itself.
Could you please re-file this bug at https://github.com/RPi-Distro/repo/issues ? (and then close this one)

alexanderpublit, 2016-04-04:
New issue: https://github.com/RPi-Distro/repo/issues/26

closed 2016-04-04 by alexanderpublit