raspberrypi/noobs #338: Please display useful information

zg, 2016-05-05:
My university requires that we have a MAC address or IP address in order to register our machines with their network.

Please provide a mechanism within NOOBS for displaying this kind of information. It would be very helpful for that intermediate step between connecting to a network and downloading the appropriate files for installation.

procount, 2016-05-05:
When in NOOBS press ctrl-alt-f2 to get to the console.
Type ifconfig to display ip and mac addresses. Press ctrl-alt-f1 to get back to NOOBS

maxnet, 2016-05-05:
Note that most operating systems that are offered through NOOBS are assuming the Pi is going to be deployed in a private home network, and have default passwords sets, and SSH enabled by default.

Especially if it happens to be one of those universities that assigns public IP-addresses to registered MAC-addresses, be aware of the security implications.

XECDesign, 2018-03-08:
Answered

closed 2018-03-08 by XECDesign