raspberrypi/noobs #339: Installation of Motion pie

Miteshdv, 2016-05-08:
This is not an issue but more kind of general query 
Can we install Motion pie on SD card with inbuilt noobs ?

I hope installing Motion pie wont erase noobs ,Please help

NicoHood, 2016-05-08:
https://github.com/raspberrypi/noobs/wiki/Adding-custom-OS-version-examples

Miteshdv, 2016-05-08:
Well ! Nichood Thanks for reply . But i am not aware how to partition SD card and it will take time to learn. I think best is to have another SD card and have Motion pie on it .

So am i right i hope it doesnt disturbs any other raspberry pi harware configuration , Pls close this ticket if i am right.

lurch, 2016-05-08:
Yes, using another SD card is *always* the easiest / most-reliable option ;-)

The Raspberry Pi doesn't have any 'internal' firmware settings, **everything** is stored on the SD card (and is configured via [config.txt](https://www.raspberrypi.org/documentation/configuration/config-txt.md)), so no matter what you do to mess up one SD card, simply putting in a known-working SD card will always bring your Pi back to the same state.

Closing this issue as requested.

closed 2016-05-08 by lurch