raspberrypi/noobs #341: NOOBS wifi preconfig

procount, 2016-05-17:
In commit  f0347db8d0eece99e97b97ab8ed32a7d584ea0c4 "Add Wfif Station Support" (oct 26 2015) a great new feature was added to copy the wpa_supplicant.conf file from the NOOBS partition to the /settings partition which allows the wifi settings for NOOBS to be preconfigured. This feature has never really been announced, although the similar feature to copy the file from the /boot partition was announced for the latest Raspbian, 
Even though I documented this feature in my update to 1.9, and announced it on the forum, there are still some people tweeting and posting that are not aware that this feature is available in NOOBS. So I think it should be announced better. **BUT BEFORE YOU DO....** there are a couple of improvements that need making.

1. In mainwindow.cpp#911 (startNetworking), the `if ( !f.exists() )` statement means that any user's wifi config file is ONLY ever copied the first time NOOBS is booted. I suspect this is to avoid copying `/etc/wpa_supplicant.conf` over the top of a user configuration on a subsequent invocation. But it prevents a user from _**ever**_ modifying the wifi config to be used by NOOBS without first deleting `/settings/wpa_supplicant.conf` (not easy for a Newbie!). So I think the logic is a bit wrong. Better to ALWAYS copy a users wifi config to /settings, and only copy the default one if it doesn't already exist 
in /settings. (Maybe even delete or rename the user's config file after copying it to avoid overwriting any manual wifi entries that are later added in the gui) That way the user can update the preconfiguration.

2. When Raspbian is installed from NOOBS, it should also look at the /settings partition for wifi configuration and install that file like it does from the /boot partition. This can be done with a simple change to `/os/Raspbian/partition_setup.sh`, as detailed in my [forum post](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=141559)

procount, 2016-05-17:
Also, if `silentinstall` is specified, startNetworking() is not executed, so `wpa_supplicant.conf` is not copied to /settings. Perhaps it should be moved outside of startNetworking()? I don't think it should be dependent upon startNetworking().

lurch, 2016-05-17:
I haven't looked at the code (and I've not done much experimentation with WiFi), but I guess it would make sense to move the copying of `wpa_supplicant.conf` outside of startNetworking() only **if** Raspbian's `partition_setup.sh` is modified to copy `wpa_supplicant.conf` from NOOBS to its own `/etc/` directory?

procount, 2016-05-17:
What's your reasoning for linking them? Isn't it easier to explain that wifi can be preconfigured in one way, rather than having lots of conditionals on it? And why make it conditional on Raspbian's modification? It's equally valid for other OSes that use a compatible wpa_supplicant.conf file.

lurch, 2016-05-18:
Sorry for the confusion - I wasn't intending to make it 'conditional', just clarifying that **until** Raspbian is modified to copy the `wpa_supplicant.conf` from NOOBS, then it 'makes sense' for NOOBS to (currently) not copy `wpa_supplicant.conf` to `/settings` unless NOOBS *itself* is using WiFi (which as you stated doesn't happen during silentinstall (and with Raspbian being the only OS included in "full NOOBS", that's the only OS that is silentinstall-able anyway) ).

Yes, other OSes could also copy `wpa_supplicant.conf` out of NOOBS, but IME they'll typically only do so after Raspbian has 'led the way' ;-)

ping @XECDesign (who I believe is the current Raspbian maintainer)

procount, 2016-05-18:
It doesn't actually require a modification to the Raspbian distro (although that may be cleaner) as it can be handled by the NOOBS `partition_setup.sh` script appropriate to each OS. So the modification can be handled entirely within NOOBS. (although I'm not exactly sure where the `os/Raspbian/partition_setup.sh` script is maintained! is it NOOBS or Raspbian?) Users can do it themselves for OSes installed on the SDcard, but it still requires a modification to the setup scripts on the download server.

lurch, 2016-05-18:
Yes, `partition_setup.sh` is exactly the modification I was referring to - apologies for causing even more confusion! I.e. in an ideal world, NOOBS copies `wpa_supplicant.conf` from the recovery partition to the settings partition, Raspbian's `partition_setup.sh` then copies it from the NOOBS settings partition to Raspbian's own boot partition, and then finally Raspbian itself copies it from its boot partition to its `/etc` directory, just as standalone-Raspbian does now. Is that right?

Even though it's NOOBS-specific, `os/Raspbian/partition_setup.sh` counts as "Raspbian" rather than "NOOBS" - this applies to all the files in the `os` directory. (although occasionally there's some slight blurring of the lines)

procount, 2016-05-18:
Yes that's right. Although I debated whether the script should copy the wpa file to /boot or straight to /etc to avoid a copy operation. i don't think it really matters which.

XECDesign, 2018-01-30:
Is 1) still an issue?

procount, 2018-01-30:
I think so, I can create a PR for how I do it in PINN if you like?
IIRC, I copy it from /mnt to /settings then rename /mnt/wpa_supplicant.conf to .bak

PR #469 (2018-10-04): Change wpa_supplicant copy
To fix #341 part 1.
Diff makes this look more complicated than it is.
Essentially the copy of wpa_supplicant.conf has been taken out of startNetwokring() into copywpa() which is then called ahead of startNetworking() regardless of silentinstall option.
Then copywpa() was modified to provide the necessary functionality.

(This PR is a copy/paste out of PINN into NOOBS)
closed 2018-10-04 by XECDesign