raspberrypi/noobs #342: Network Cache

procount, 2016-05-19:
QUESTION: In my fork, I download some additional files when connected to the network. If I then reboot into "noobs" without the network, it appears that the file is still downloadable, whereas the standard downloadable files are not. This appears to be due to the network cache, because if I delete the cache folder on /settings, the file is no longer available. Why is the behaviour different for these files?
Does the download server have some special settings to stop NOOBS from caching the files?
Is the network cache only there to improve performance during a single noobs session? In which case, can I safely delete it on startup?

lurch, 2016-05-19:
Can you give examples of files (and their corresponding URLs) which do and don't get cached?

procount, 2016-05-19:
For example: http://downloads.raspberrypi.org/os_list_v3.json does not get cached. but 
 https://raw.githubusercontent.com/procount/pinn-os/master/os/os_list_v3.json does.
(Like I said, this is only related to my fork, not noobs itself)

lurch, 2016-05-19:
> Why is the behaviour different for these files?

I've just had a rummage, and it looks like https://github.com/raspberrypi/noobs/blob/11ba0b51db6fa48c85b5c8cd13ca757f64a6bb96/recovery/mainwindow.cpp#L961 is where the cache gets initialised. I can't see any "special behaviour" set up, so I guess the difference must be down to the HTTP headers?

```
$ curl --head http://downloads.raspberrypi.org/os_list_v3.json
HTTP/1.1 200 OK
Date: Thu, 19 May 2016 12:35:26 GMT
Server: Apache/2.2.22 (Debian)
Last-Modified: Fri, 13 May 2016 08:25:14 GMT
ETag: "6008-248e-532b503932e80"
Accept-Ranges: bytes
Content-Length: 9358
Content-Type: application/json

$ curl --head https://raw.githubusercontent.com/procount/pinn-os/master/os/os_list_v3.json
HTTP/1.1 200 OK
Content-Security-Policy: default-src 'none'; style-src 'unsafe-inline'
X-XSS-Protection: 1; mode=block
X-Frame-Options: deny
X-Content-Type-Options: nosniff
Strict-Transport-Security: max-age=31536000
ETag: "ac238cf4e141b02dcfd80ea7e4038ba22ed7e808"
Content-Type: text/plain; charset=utf-8
Cache-Control: max-age=300
X-GitHub-Request-Id: 17EB2B16:7A84:5CFCB24:573DB320
Content-Length: 2440
Accept-Ranges: bytes
Date: Thu, 19 May 2016 12:35:45 GMT
Via: 1.1 varnish
Connection: keep-alive
X-Served-By: cache-ams4145-AMS
X-Cache: MISS
X-Cache-Hits: 0
Vary: Authorization,Accept-Encoding
Access-Control-Allow-Origin: *
X-Fastly-Request-ID: b657755770f084215ee6856833a78a3ba48967a0
Expires: Thu, 19 May 2016 12:40:45 GMT
Source-Age: 0
```

> Is the network cache only there to improve performance during a single noobs session? In which case, can I safely delete it on startup?

The cache is obviously empty the first time NOOBS runs, so maybe in your fork you could use http://doc.qt.io/qt-4.8/qnetworkdiskcache.html#clear ?

procount, 2016-05-19:
Ahh! probably the pesky `Cache-Control: max-age=300` does it.
Thanks. I'll give it a go.

procount, 2016-05-19:
Sorted! thanks

lurch, 2016-05-19:
Okay to close?

closed 2016-05-19 by procount