raspberrypi/noobs #343: (feature request) allow installation from USB

altsheets, 2016-06-04:
When starting NOOBS, I only see very few options for (Raspian, 2 media centres, Windows, ...) - and all of them need to be downloaded from the internet. We have a VERY very slow internet collection.  

Plus, I would like to install two customized Raspian versions called ROKOS (one for altcoins one for bitcoin) - see [this](https://bitcointalk.org/index.php?topic=1084908.msg15068067#msg15068067) and [this](https://bitcointalk.org/index.php?topic=1084908.msg15081716#msg15081716) post if you want to.

_Feature request:_  
I would much appreciate **NOOBS to be able to use a local data repository, e.g. a USB-stick.** (And I would imagine it is not too difficult to add a button/option "install from local path").

Secondly ... what is the easiest way, to include 2 such "custom images", from within, or into NOOBS? (the images I would use are [here](http://rokos.space/downloads.html)). Is it very complicated? Does it need the image maker (rokos.space) to change much about his image? 

Any hint much appreciated! Thank you very much.

procount, 2016-06-05:
There are detailed instructions on how to create your own custom inages for noobs in the documentation see README.md at https://github.com/raspberrypi/noobs

If you want to install from USB stick then use PINN which is a fork of noobs that supports that, plus installing from another internet source and other features. See https://github.com/procount/pinn

altsheets, 2016-06-05:
Wow, thanks a lot. That is great! Many many thanks.

Jimmy-Z, 2016-12-19:
nice, I'll try that later.

procount, 2017-11-03:
Now included - close?

XECDesign, 2017-11-03:
My read of it is that @altsheets wants the os/ directory to be read from USB, but NOOBS and OS itself to be installed to SD (or another USB MSD).

procount, 2017-11-03:
Yes I agree. And it's done already isn't it - usbaddimages() or something like that?

XECDesign, 2017-11-03:
Ah, if that's the case, great. If not, it would be a wontfix anyway, so it should be closed either way. Thanks

closed 2017-11-03 by XECDesign

procount, 2017-11-03:
mainwindow::addImagesFromUSB()
https://github.com/raspberrypi/noobs/blob/master/recovery/mainwindow.cpp#L1755 
:smile:

XECDesign, 2017-11-03:
Nice, thanks.

altsheets, 2017-11-10:
Great, thanks.

I have now told the creator of that ROKOS image:

https://bitcointalk.org/index.php?topic=1084908.msg24326674#msg24326674