raspberrypi/noobs #348: URL redirection help

procount, 2016-07-15:
I am trying to get NOOBS to download some files from sourceforge.net but the redirection does not work. I can download a file from there using wget without issue. It goes through about 5 levels of redirection using http codes 307, 301,302 and finally 200. But when I try the same thing with NOOBS downloadmetafile() and similar functions, the first httpstatuscode is 0 and there is no redirectionurl and no payload. I have tried various urls to try and get it to work. The only url that does work is a fully resolved mirror URL (from the final 200 code response above), but they are not persistent and it is not recommended to use those, but to go through their mirrors for load balancing etc. etc.
Is anyone familiar with the vagueries of Sourceforge redirection ? (@Maxnet - do you have any experience with this?)

maxnet, 2016-07-15:
Don't know.
I use libcurl in my own project that uses SF for hosting, and that takes care of the redirects transparently.

If HTTP status code is 0 check for network (or SSL if applicable) errors.

procount, 2016-07-16:
I already checked for SSLerrors, but a signal was never raised.
So I have now checked for Network errors and I've got a QNetworkError::RemoteHostClosedError(2).
It's a bit tricky to debug SSL messages in Wireshark, but I get 2 hello messages, certificate, server key exchange, (client key exchange, change cipher spec, Encrypted Handshake message), Change cipher spec, Encrypted handshake message, Application Data then a reset. I'm not really familiar with this, but it looks like the SSL key exchange takes place ok, but the server does not like the Application Data...?
So I tried again with a HTTP only request and got another reset. Checking the headers against the same request from Firefox, and the Accept: header was missing. I added this, but it is still not accepted.

procount, 2016-07-19:
Finally fixed it!
Sourceforge did not like the User-Agent for some reason. After replacing it, it works fine.

closed 2016-07-19 by procount