raspberrypi/noobs #349: ACT LED should blink for some time during start of NOOBS 1.9.2

ComputerDoktor, 2016-07-31:
I was ready to return my new RPI3 as defect, when nothing happened at power-on except PWR LED on and one short flash of ACT. According to 
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=58151
this indicates that the SD-card could not be read.
Only later on I found out that NOOBS had modified the partition table and that the HDMI-to-VGA adapter was failing.

procount, 2016-07-31:
IIRC, NOOBS will cause the ACT LED to come on and flash intermittently the FIRST time you use it on an Sd card. During this time it is accessing the SD card a lot to partition and format it. If you have problems with your display it is easy to miss this behaviour.

However, once NOOBS has successfully re-partitioned the SD card, subsequent reboots are very quick. The linux root file system is all loaded v.quickly into a temporary RAM disk so after that there are no disk accesses until you choose to install an OS. You will just see one quick flash of the ACT LED..

Errors reading the SD card are normally indicated by a regular flashing of the LED where you can count the regular flashes to indicate the error code.

I don't think it is necessary for NOOBS to blink the ACT LED anymore than it does. The sticky thread is also not wrong, but perhaps it needs this additional info added for NOOBS. The 20 secs it talks about is for a standard distro like Raspbian to load. NOOBS takes <1 sec in disk access on subsequent boots.

ComputerDoktor, 2016-07-31:
I have just reformatted the SD-card (class 10) with 32 GB FAT32 and copied NOOBS 1.9.2 onto it. 
I could not tell any difference between the first and second boot of NOOBS on my RPI3: one short flash of ACT and then dark. It appears to me that ACT does not blink during the re-partitioning.
I think letting ACT blink for maybe 5 seconds could help to prevent a few unnecessary returns of the hardware.

maxnet, 2016-07-31:
Probably related to #347
Think our /sys/class/leds went missing after a kernel/firmware upgrade. Not sure why, might need to enable a different kernel option.

procount, 2016-07-31:
@ComputerDoktor - Indeed, the ACT LED seems to have stopped working on the Pi3.

@maxnet - I seem to recall the ACT LED is driven differently on the RPi3 - from a GPU port expander? Maybe this is the issue?

procount, 2016-08-03:
If it helps, the following log message is displayed on RPi3 when NOOBS boots:
`Can't create /sys/class/leds/led0/trigger: nonexistent directory`

Not sure about NOOBS, but on PINN I also get:
'brcmfmac: brcmf_sdio_drivedtrengthinit: No SDIO Drive strength init done for 43430 rev 1 pmurev 24`

maxnet, 2016-08-04:
>@maxnet - I seem to recall the ACT LED is driven differently on the RPi3 - from a GPU port expander? 

Ah, seems to just need ``CONFIG_GPIO_BCM_VIRT=y``

procount, 2016-08-04:
Yep, that seems to fix it on the RPi3. Seems to still work on PiZero too.

procount, 2016-10-05:
@XECDesign - Did you miss this little fix for RPi3 in the V2.0 release...?
`CONFIG_GPIO_BCM_VIRT=y`

XECDesign, 2016-10-05:
I don't maintain NOOBS, so I am not aware of the issues with it. This was only a quick update to get it working on the Pi 2+. If I'm doing the next update, I've added this fix to the todo list.

commit 854be817 (2016-10-30): Pi 3: fix ACT led

Add missing CONFIG_GPIO_BCM_VIRT=y kernel config option.

Closes #347
Closes #349
closed 2016-10-30 by maxnet