raspberrypi/noobs #350: Can't enable RTL8188CUS wifi driver

procount, 2016-08-01:
Having successfully added wifi drivers for RTL5370, zd1211rw, mt7601u and some Ethernet drivers like DM9601, I am really struggling to add the driver for the RTL8188CUS chipset (made more difficult because I don't have such a wifi dongle). 
Here is the output from Raspbian Dmesg where it works:
[    3.992386] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=8176
[    4.005040] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.017985] usb 1-1.3: Product: 802.11n WLAN Adapter
[    4.028545] usb 1-1.3: Manufacturer: Realtek
and lsmod (relevant excerpt):
8192cu                597488  0 
cfg80211              499234  1 8192cu
rfkill                 21397  4 cfg80211,bluetooth

It appears to use the 8192 driver so I have enabled the kernel option:
Device Drivers > Network device support > Wireless LAN > Realtek 8192C USB WiFi.
Networking support > Wireless > cfg80211 - wireless configuration API is already enabled.

and in menuconfig I enabled:
Target packages > Hardware handling > Firmware > WiFi firmware > Realtek 81xx 

In the NOOBS dmesg output, the driver seems to be available:
[    0.435165] usbcore: registered new interface driver zd1211rw
[    0.435320] usbcore: registered new interface driver rtl8187
[    0.435438] usbcore: registered new interface driver rtl8192cu
[    0.435564] usbcore: registered new interface driver rt2500usb
[    0.435689] usbcore: registered new interface driver rt73usb
[    0.435815] usbcore: registered new interface driver rt2800usb
[    0.435941] usbcore: registered new interface driver mt7601u
[    0.436461] usbcore: registered new interface driver brcmfmac
[    0.436650] usbcore: registered new interface driver rtl8150
[    0.436783] usbcore: registered new interface driver r8152
[    0.436899] usbcore: registered new interface driver asix
[    0.437015] usbcore: registered new interface driver ax88179_178a
[    0.437233] usbcore: registered new interface driver dm9601
[    0.437371] usbcore: registered new interface driver smsc95xx

and the USB device is detected:
[    2.172954] usb 1-1.3: new high-speed USB device number 4 using dwc_otg
[    2.274470] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=8176
[    2.274500] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.274514] usb 1-1.3: Product: 802.11n WLAN Adapter
[    2.274525] usb 1-1.3: Manufacturer: Realtek

Is it significant that CRDA fails?
[   34.942895] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA

But wifi dongle is not enabled. Any ideas what I am missing?

procount, 2016-08-04:
FInally, I managed to get this driver working and wlan0 is now present.
But the recovery program seems to have trouble configuring it. Here is an extract from debug.txt:

QDBusObjectPath: invalid path ""
Using wifi interface "" 
QDBusObjectPath: invalid path ""

Could it be this driver is not compatible with cfg80211, or dbus? (Not so easy to debug without having the wifi dongle itself!) 
I need to look at the configuration in a bit more detail - anything I should be looking out for?
What should I do - fix the driver, or modify how it is configured?

maxnet, 2016-08-04:
>Could it be this driver is not compatible with cfg80211, or dbus? (Not so easy to debug without having the wifi dongle itself!) 

Would argue that if you do not have the dongle yourself, you should let whoever wants support for it, fix it.

That said, if the driver does not speak nl80211, but only legacy wext, wpa_supplicant should be started with the right flag to tell it to speak that.
Since wpa_supplicant is started indirectly by dhcpcd in the current implementation, look in the documentation of that for options.

procount, 2016-08-04:
> you should let whoever wants support for it, fix it.

If only! But he's a user not a developer. And I think the 8188 chipset is quite common, so I'd like to get it working. In any case, it is another learning opportunity for me, since dbus and wpa_supplicant innards are all new to me.
Thanks for the pointers - I'll start looking in that direction.

procount, 2016-08-04:
IIRC I read that the 8188 chipset didn't work with the standard hostapd, and there is a special build of hostapd that needs to be used. Could that be because of the same issue?

maxnet, 2016-08-04:
Could be.
Recall standard wext does not support AP mode.

procount, 2016-09-16:
JFYI: I got the 8188cus chipset working and wpa_supplicant will configure it with an existing conf file. But selecting an SSID and configuring the wifi needs an alternative approach to dbus/nl80211. Currently I'm using wpa_cli which works most of the time, but I have to fix a pesky seg fault that *sometimes* occurs.

I found that I had to launch wpa_supplicant myself, since dhcpcd wasn't starting it, or at least not with the correct parameters. Probably because it couldn't find its interface, even though I added wext.

procount, 2016-09-21:
I discovered that some work has been done to update the 8188cus driver to use the cfg80211 interface in
raspberrypi/linux#1627 also 1624, 1488 & 1489 If I update the kernel to 4.8, would this solve the whole issue of this driver?

procount, 2016-10-05:
So I back-ported the cfg80211 version of 8192C to noobs and it seems to work. BUT.....
I now want to add the driver for the 8188EU chipset.
This is available as a staging driver. but when I enable it, there seems to be many duplicate definitions with the 8192C driver. How can I get these 2 drivers built-in to the kernel without such conflicts?
There seems to be some external repositories for different and experimental versions of the Realtek wifi drivers but I'm really getting lost as to which ones to use.
I even tried bumping the kernel to 4.8, but the same conflicts arise. I guess building the drivers as kernel modules avoids such name clashes.
Any ideas?

procount, 2016-10-20:
Solved.

closed 2016-10-20 by procount

Merlinstail, 2017-08-18:
I'm looking to resolve this in Arch Linux too.  How did you get around the problem?  For some reason it seems to load the rtl8192cu driver which crashes and is out of date.  Realtek's website no longer has a linux driver download so I'm stuck

procount, 2017-08-18:
See https://github.com/lwfinger/rtl8188eu/issues/185