raspberrypi/noobs #354: Serial port baud rate for NOOBS.

kowoba, 2016-09-01:
I have a need to run my serial consoles at other speeds than 57600 and 38400, and struggle with the systems on the other end not treating it well when the RPi boots and NOOBS' sends its "Welcome to the rescue system recovery login.... Uncompressing Linux... done, booting the kernel." at 115200 baud. This comes over as garble and causes random reactions on the systems to which the Pi is hooked onto.

I tried to set it by setting init_uart_baud=57600 in /boot/config.txt, but that does not help.

By using a terminal set to 115200 baud, I managed to quickly log in with root/raspberry and paste in a "cat /proc/cmdline" and saw all the paramters used, none of them saying anything about baud rate, so after a little more googling, I found that NOOBS' kernel has the baud rate is set in the kernel config, without any other options available:
https://github.com/raspberrypi/noobs/blob/b7d838d5b0cc2db436c5ca1265d9999bb5b53b85/buildroot/.config#L286

So I find myself in a bit of a sucky situation - do I need to rebuild NOOBS every time I have a system with a different baud setting? Would be be possible to have the NOOBS kernel take into account what the value of init_uart_baud is, instead of using a "hard coded" 115200? Is there perhaps a cmdline.txt kind of file that the NOOBS' kernel can read on boot?

lurch, 2016-09-02:
In all the years that NOOBS has been released, you're the first person to ask about the serial baud rate.
So I suspect that for now you'll be stuck with compiling your own kernel :-/

maxnet, 2016-09-02:
Do you really have a need for serial console in NOOBS itself in the first place, or is the problem the garbage sent?

If the latter try killing it instead, by renaming the recovery files to their normal equivalents (e.g. recovery.elf -> start.elf) and using a config.txt among the lines of:

```
initramfs recovery.rfs
dtparam=uart0=off,uart1=off
```

kowoba, 2016-09-02:
Yes - I suspected that I was pretty much alone on this issue :)

My problem is indeerd the garble it generates, so "skipping" the recovery part all-together is of course what I should do! Thanks for the tip, I will try later today!

kowoba, 2016-09-04:
I have no tried various, but I haven't managed to get a "57600 baud clean" boot yet. I have managed to disable serial output all-together, but that is not what I want - I need the null-modem connection.

`
root@mist:/boot# ls -lrt
total 20828
-rwxr-xr-x 1 root root     294 Dec 31  1979 os_config.json
-rwxr-xr-x 1 root root     128 Jan  1  2000 cmdline.txt
-rwxr-xr-x 1 root root   18974 May 10 23:50 LICENSE.oracle
-rwxr-xr-x 1 root root     145 May 10 23:50 issue.txt
-rwxr-xr-x 1 root root   13484 Jun 30 18:13 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x 1 root root   14567 Jun 30 18:13 bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 root root   13257 Jun 30 18:13 bcm2708-rpi-cm.dtb
-rwxr-xr-x 1 root root   13221 Jun 30 18:13 bcm2708-rpi-b.dtb
-rwxr-xr-x 1 root root 4123112 Jun 30 18:13 kernel.img
-rwxr-xr-x 1 root root   15237 Jun 30 18:13 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root 4224232 Jun 30 18:13 kernel7.img
-rwxr-xr-x 1 root root   18693 Jun 30 18:14 COPYING.linux
drwxr-xr-x 2 root root    8192 Jun 30 18:14 overlays
-rwxr-xr-x 1 root root    9717 Jun 30 18:15 fixup_db.dat
-rwxr-xr-x 1 root root    6482 Jun 30 18:15 fixup.dat
-rwxr-xr-x 1 root root    2504 Jun 30 18:15 fixup_cd.dat
-rwxr-xr-x 1 root root    1494 Jun 30 18:15 LICENCE.broadcom
-rwxr-xr-x 1 root root    9717 Jun 30 18:15 fixup_x.dat
-rwxr-xr-x 1 root root   17932 Jun 30 18:15 bootcode.bin
-rwxr-xr-x 1 root root  617432 Jul 25 09:27 start_cd.elf
-rwxr-xr-x 1 root root 4926264 Jul 25 09:28 start_db.elf
-rwxr-xr-x 1 root root 3877720 Jul 25 09:29 start_x.elf
-rwxr-xr-x 1 root root 2746552 Sep  2 22:43 start.elf
-rwxr-xr-x 1 root root    1793 Sep  2 23:00 config.txt
root@mist:/boot#
`

This is my /boot - from what I understand, the start_db.elf and start_x.elf are "debug" and "extra codecs", none of them really interesting for me and my rpi zeros. I have rename start_cd.elf as start.elf just to see how that works, and it works just fine, I cannot tell any difference. I understand kernel.img is for RPi1 and kernel7.img is for RPi2 - my Zero clearly boots kernel.img.

From what I understand, the Pi initially boots from GPU that loads bootcode.bin, that sets up memory and loads start.elf with parameters from config.txt, and then fires up the CPU to run from start.elf, start.elf then bootstraps the OS by loading the kernel.img and feeding it parameters from cmdline.txt along with various "hard coded" boot parameters. What I don't understand, is where NOOBS is located in this picture. Inside the start.elf? Within bootcode.bin? I don't have any recovery.elf or recovery.img here (which I gather would be used it start.elf was missing). I clearly see two loadings of a Linux kernel - one "recovery mode" (115200 baud), and one "final" mode (using cmdline,txt, respecting 57600 baud boot params). This is raspbian, so maybe it is the very same kernel that is loaded twice? First initially to a "recovery" ramdisk without use of cmdline.txt, and secondly (by means of kexec I guess) after reading, parsing and establishing that cmdline.txt makes sense and root filesystem will be found? I am just speculating here, guess I should take this up with Raspbian people instead :)

maxnet, 2016-09-04:
>What I don't understand, is where NOOBS is located in this picture

NOOBS files do not life inside Raspbian's /boot, but in their own hidden recovery partition.

```
sudo bash
mkdir /mnt/noobsfiles
mount /dev/mmcblk0p1 /mnt/noobsfiles
```

Mess with the files inside /mnt/noobsfiles, ``umount /mnt/noobsfiles`` when done.
E.g. turn uart off for NOOBS there like I suggested (this requires renaming recovery files to their normal equivalents, otherwise the contents of config.txt is ignored).
Or disable NOOBS by creating an ``autoboot.txt`` file with ``boot_partition=6`` into it there (assuming only one OS is installed)

lurch, 2016-09-04:
It's a bit out of date now, but there's some info about NOOBS's partitioning on the wiki https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained

aodonnell-gpfw, 2016-11-10:
seems to me that I have a similar issue.  I have the Noobs SD card rev 1.8 and am running on a Pi 3 model B, out of the box the serial port outputs at most 20 sensible characters, then looks like a timing error and unintelligible.  I used node red to make a simple loop back test with external loop back cable and what goes out comes back cleanly.  It's almost like the clock is marginal and so getting bit alignment issues.  Its definitely code/config dependent as installing todays Raspbian allowed it all to work.  Installing Noobs 2.0 onto an SD card, hope this fixes my isses.

lurch, 2016-11-10:
To get reliable UART output on a Pi 3B, you need to add `enable_uart=1` to `config.txt`. See [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=141195) for more info.

closed 2018-01-30 by XECDesign