raspberrypi/noobs #355: Developer question about NOOB extension

steilerDev, 2016-09-14:
Hello guys,

thank you for the great job with NOOBS. I want to build upon your system, enabling remote provisioning of Raspberry Pi's. Therefore I would like to rewrite the recovery application and add new modules. However this project is a Makefile nightmare (especially for someone like me who has only basic knowledge of Makefiles) and I can't find the location, where I can strip out and include new source files for the recovery application.

Could you help me here and point me in the right direction?

Thanks!

ghollingworth, 2016-09-14:
The Recovery application is a QT application so the project file is:

https://github.com/raspberrypi/noobs/blob/master/recovery/recovery.pro

Thanks

Gordon

closed 2016-09-14 by ghollingworth

lurch, 2016-09-14:
You might want to get in touch with @procount who's developed https://github.com/procount/pinn

steilerDev, 2016-09-14:
@ghollingworth Thanks for that. (never used a qt application, I've got a lot of work in front of me ^^)
@lurch I will have a look, thank you!

lurch, 2016-09-14:
Depending on what you're trying to do, I wonder if [netbooting](https://www.raspberrypi.org/blog/pi-3-booting-part-ii-ethernet-all-the-awesome/) might suit you better?

steilerDev, 2016-09-14:
@lurch Looks great, but only available for Pi3, we want to be able to address all models, but I will monitor that feature, thanks!