raspberrypi/noobs #364: NOOBS translation issues - rebuild language file project

bovirus, 2016-12-18:
Hi.

@lurch 

I verifeid that the NOOBS is a good project but there many problems about localizations.
The lanugage files include has several strings missng

DE - 42 of 146
FI - 42 of 146
FR - 25 of 146
HU - 113 of 146
JA - 21 of 146
NL - 99 of 146
PT - 42 of 146
RU - 123 of 126

Then as you can find the multilanguage is quite back to teh final destination.
Then running lupdate the language has a duplicate strings

Then there area several langauges prosed in teh apst but never added to tehr epository

  
  Asturian =  by softastur - 26.07.2015
  Basque = by aldatsa - 08.02.2105
  catalan = by cescp - 24.02.2014
  French = by qsypoq - 25.01.2016
  Dutch =  by ppigmans  - 25.03.2016
  German = by fonic - 13.03.2016
  Italian = by dforsi - 01.09.2015
  Italian =  by CereX-ITA  - 10.03.2014
  Italian = by bovirus 13.12.2016
  Korean = by sople1 - 29.06.2014
  Japanes = by Akkiesoft - 24.02.2014
  polish = by kadamski  - 28.12.2015
  Spanish = by ResonantWave - 08.02.2016
  Spanish = by amorellgarcia - 25.02.2014
  Swedish = by Mangero - 22.03.2015
  Turkish = by diorray - 27.09.2016
  Turkish = by oguzhank77  - 15.01.2016
  zh_tw = by s8321414 - 24.02.16
  
Why didn't yopu implement all languages in your build?

@softastur
@aldatsa
@cescp
@qsypoq
@qsypoq
@fonic
@sople1
@Akkiesoft
@kadamski
@ResonantWave
@Mangero
@diorray
@s8321414

Coudl you please send me your latest version of your language file (langauge_xx.ts where xx is your language) to bovirus at gmail.com (i used space to avoid spam search)?

I will develop a special build with all languages available.

Thanks.

s8321414, 2016-12-18:
OK, I will do it if I have free time recently.

bovirus, 2016-12-18:
@s8321414
Please don't edit it now.
Please send me the last file that you uploaded/have.
I will apply all latest changes that was made by master English file and I will send back for final translation.
You can also attach it to the message here.

s8321414, 2016-12-19:
@bovirus Excuse me, but how can I re-use my previous translation and sync it with new translation template on Qt's .ts format?

aldatsa, 2016-12-19:
@s8321414 You can sync your previous translation with the new template with this command:

`lupdate recovery/recovery.pro`

aldatsa, 2016-12-19:
I updated my translation (#229) to include the newest strings

s8321414, 2016-12-20:
hmm, seems I also need `-ts translation_zh_TW.ts` option to sync my translation, but it says 0 new string for mw...

bovirus, 2016-12-20:
@s8321414

Could you please attach to the message your translation_zh_TW.ts language file?
The strings are1 46.

@aldatsa
the command to update translation and discard obsolete strings is 
lupdate -no-obsolete recovery/recovery.pro
Could you please attach to the message your basque language file?

s8321414, 2016-12-20:
[translation_zh_TW.zip](https://github.com/raspberrypi/noobs/files/663589/translation_zh_TW.zip)
hmm, GitHub don't want me upload .ts file, so I upload a zip...

aldatsa, 2016-12-20:
This is strange. I tried to upload a zip file but github says that they don't support that file type. So I uploaded a gz file:
[translation_eu.gz](https://github.com/raspberrypi/noobs/files/663615/translation_eu.gz)

bovirus, 2016-12-20:
@s8321414 
Thanks.
GitHub accept zip file (not .ts)

@aldatsa
Thanks for the file
It's very strange.

I will update the file in first post and I will rebuild NOOBS.

s8321414, 2016-12-20:
@bovirus you may need zh_TW language flag from here: https://github.com/l10n-tw/noobs/blob/master/recovery/icons/zh_TW.png

aldatsa, 2016-12-20:
@bovirus You can use this flag for the Basque language: https://github.com/aldatsa/noobs/blob/i18n-eu/recovery/icons/eu.png

bovirus, 2016-12-20:
@Akkiesoft

Please upload the file here (as .zip file) or upload it in one host service and post here the link to download the full translation file.

bovirus, 2016-12-22:
@aldatsa 
@Akkiesoft
@s8321414 

In the first post you can find

- latest build of all languages available

- a new build of NOOBS with all languages included.
Please take care that if you wan to change default language/keyboard change the recovery.cmdline file.

XECDesign, 2017-01-05:
Thanks @bovirus, that's very helpful. When you're happy with it and if @maxnet doesn't see any problems with it, we can make sure it's picked up in the next release.

bovirus, 2017-01-05:
@XECDesign 

Please include in NOOBS next release not only Italian but all languages explained in first post (please sue my language files that are already checked).

XECDesign, 2017-01-05:
Of course. Do you intend to add further updates or do you think everybody who was going to update their translations already has?

bovirus, 2017-01-05:
@XCDesign
For me Italian is OK (complete and tested).
For teh other languages deopned by other users.
If you agree I want to collaborate with you to handle the languages.

XECDesign, 2017-01-05:
Yes, but I wouldn't be of much help other than to merge them when they're ready. I would rather do it as a batch rather than keep adding individual languages and updates every few days, if that's possible.

bovirus, 2017-01-06:
XECDesign

To help the translator and otehr user please add the trings (commneetd) in recovery.cmdline file

like

# lang=<two-letter language code> (e.g. lang=de or lang=en)

# keyboard=<two-letter layout code> (e.g. keyboard=de or keyboard=us)

# display=<display mode number> (e.g. display=1 or display=3)

# partition=<partition_number> (e.g. partition=6)

# showall (shows all available OSes regardless of your Raspberry Pi model)

maxnet, 2017-01-07:
> NOOBS multilanguage - R04: https://drive.google.com/open?id=0ByL-fx0GbfuicGczcldNWS1Tc2M

You are missing the Korean font in your build.

I also suggest that translators try out their language, and verify that it looks ok visually.
Can spot several languages where the translator did translate every string, however the translatated words are much longer than the original ones, and simply do not fit correctly in the user interface.
It could be necessary to use less exact translations (e.g. "wifi networks" -> "wifi" or "networks"), or abbreviations (like we have "edit config" in English rather than "edit configuration") if that is what it takes to make all buttons fit in the toolbar of the main window.

>To help the translator and other users please the strings (commented) in default recovery.cmdline file

That is not possible given there can only be a single line in that file.

aldatsa, 2017-01-07:
I made some small corrections to the Basque translation after testing it using @bovirus's build. I updated my pull request (#229) and I attach the updated translation file.
[translation_eu.zip](https://github.com/raspberrypi/noobs/files/691576/translation_eu.zip)

bovirus, 2017-01-07:
@maxnet 
Sorry but I didn't undersatnd what's the problem.
I got the Korean file directly by pull requesta nd copied it witha lls pecial (Korean characters).
Then pelase make more details baout it or please post the right Korean file.

Please take care that translator can check their tarnslation only if thier translation are in place in a development build with all languages.

@aldatsa 
Thanks for the update.
I will update it in the build next Monday.

maxnet, 2017-01-07:
>I got the Korean file directly by pull requesta nd copied it witha lls pecial (Korean characters).

Looks like this to me:

[image]

Make sure you do not just copy the .ts file, but apply all changes in the PR including the ones that cause the extra font file to be added: https://github.com/raspberrypi/noobs/pull/199/files

One way to do that is to download all changes as .patch file: https://github.com/raspberrypi/noobs/pull/199.patch
And apply that to your local tree with patch or "git am"

bovirus, 2017-01-08:
@maxnet 
Thanks for the info & the patch.
I will add to my build.

@XECDesign 
For the rebuild pelase 
- use latest (R05) language builds (improve eu/it language)
- include patch for korean langauge by maxnet
- check the langaueg file ad double line 44 (problem posted some posts ago)
Thanks.

aldatsa, 2017-01-08:
@bovirus 
There are two translatable strings in the line 44 of the file wifisettingsdialog.cpp:

44 `QMessageBox::critical(parent, tr("No wifi interfaces"), tr("No wifi interfaces available"), QMessageBox::Close);`

https://github.com/aldatsa/noobs/blob/i18n-eu/recovery/wifisettingsdialog.cpp#L44

bovirus, 2017-01-08:
@aldatsa 

When I applied the alignment translation tool (lupdate) sometimes in some languages update it to  146 strings and on other langauge udate it using master English at 147 strings.

aldatsa, 2017-01-08:
@bovirus 
Yes, you are right. I noticed the difference but I don't know the reason.

bovirus, 2017-01-08:
@aldatsa 

I believe that the develop should dientify the two strings for lien 44 with two different lines.

aldatsa, 2017-01-08:
@bovirus 
I don't think that's the cause of the difference in the number of strings. There are more files with two translatable strings in the same line. For example:

150 `QMessageBox::critical(this, tr("Reformat failed"), tr("SD card might be damaged"), QMessageBox::Close);`
https://github.com/raspberrypi/noobs/blob/master/recovery/mainwindow.cpp#L150

234 `QMessageBox::critical(this, tr("Connecting to wifi failed"), tr("Connecting to the wifi access point failed. Check your password"), QMessageBox::Close);`
https://github.com/raspberrypi/noobs/blob/master/recovery/wifisettingsdialog.cpp#L234

44 `QMessageBox::critical(this, tr("Cannot display boot menu"), tr("Error mounting settings partition"));`
https://github.com/raspberrypi/noobs/blob/master/recovery/bootselectiondialog.cpp#L44

maxnet, 2017-01-10:
>I believe that the develop should dientify the two strings for lien 44 with two different lines.

Line numbers shouldn't matter.
Maybe lupdate decided the strings are so similar that they are identical though.
May be able to override that by running it with ``-disable-heuristic similartext``

XECDesign, 2017-01-13:
@bovirus I guess there was a miscommunication somewhere. I wrote that the translations will be picked up in the next release, not that there will be a release in the next few days. In general, there is a new release when there are critical bug fixes or major features added. A new release involves masteing a new SD card image for distributors, a period of testing and then re-flashing a lot of SD cards (the ones sold with NOOBS and Raspbian pre-installed).

So, before the next release comes around, I will go through all of the translations and some of them will be picked up.  The ones which have issues which haven't been addressed will be closed as inactive. I don't think there will be a release any time soon, so that should give plenty of time for issues to be ironed out. Aside from that, there is not much more I can do. Translations are 100% a community contribution, the best I can do is merge them. I am not a NOOBS developer, I don't add new features or say what goes in and what doesn't.

bovirus, 2017-01-13:
@XECDesign

I believe that the NOOBS developers ignore for a long time this project and the translations posted by users (without replying them - not accettable).
The only way for the devloper to say thanks to all you guys that want to contribute with translations I believe that is don't wait again for the new version (its sound as en excuse) with all languages in place.

You should relaese a new version to include all languages and include Korean fonts.

I believe that is only way to get back credibility in this project by the guys that partecipating to the frum and posted translations after that for mroe than two years no one of teh dvelopers anwsered to all guys thar want to contribute levaing them without any feedbacks.
 
I just do it in very short time (first post image vailable). 

**If you want you can do it.**

ghollingworth, 2017-01-13:
@bovirus Your comments are not applicable to this issue, if you want to open an issue concerning our release schedule then you are welcome to do that.  Any further complaints about the way this is done will be deleted from this issue

bovirus, 2017-01-16:
@XECDesign 

I made a pull-up request with all changes request for new languages.

https://github.com/raspberrypi/noobs/pull/373

Thanks.

procount, 2017-01-31:
In order to translate the standard dialog boxes with Yes, No Cancel etc. button text, is it necessary to include the qt_xx.qm files as well? The qt_xx.ts files exist for a lot of the languages, but some are missing.

bovirus, 2017-01-31:
@procount 
I can prepare it the file for Italian language hjust for Yes/NO/etc (it will be more or less 20 strings).
The other translators can made the same for other languages basing on master English that I will prepare.

maxnet, 2017-01-31:
>In order to translate the standard dialog boxes with Yes, No Cancel etc. button text, is it necessary to 
>include the qt_xx.qm files as well?

Recall we already had some code to add some standard sentences to the strings to be translated for languages that do not have qt_xx.qm

https://github.com/raspberrypi/noobs/blob/7854f404d7653233f0d4c4d729ceed1e10ab9e88/recovery/languagedialog.cpp#L25

bovirus, 2017-01-31:
I believ that is not necessary.If you run 

lupdate recovery/recovery.pro -no-obsolete -disable-heuristic similartext

(tested) the lupdate comamnd add the relative message of dialog box in translate_xx.ts language file.
You can edit it and fill-in the 8 strings missing.

procount, 2017-01-31:
Ok. I assumed those translations weren't being picked up due to previous comments. Maybe the code to load the additional qt*.qm files is not necessary then.

XECDesign, 2017-02-28:
I've pulled in commits from your repo, thanks.
I've noticed that this has created two versions of 'Spanish', the other one being Asturian. Was that intended?

closed 2017-11-03 by XECDesign