raspberrypi/noobs #365: With version V2.1,  You can't boot SSH headless.  The instructions provided don't match the software.

zipzit, 2016-12-26:
Goal: 
First time boot up of a system that can be managed headless (without keyboard, mouse or display), via SSH

Background:
There have been significant changes to the "New Out Of the Box Software" (NOOBS) system with [version 2.1 . ](https://www.raspberrypi.org/downloads/noobs/)  According to the release notes included with the package:

> 2016-11-25: * SSH disabled by default; can be enabled by creating a file with name "ssh" in boot partition

[There has been some pretty good discussion on the reasoning for the change,](https://www.raspberrypi.org/blog/a-security-update-for-raspbian-pixel/) where SSH is involved.

> The boot partition on a Pi should be accessible from any machine with an SD card reader, on Windows, Mac, or Linux. If you want to enable SSH, all you need to do is to put a file called ssh in the /boot/ directory. The contents of the file don’t matter: it can contain any text you like, or even nothing at all. When the Pi boots, it looks for this file; if it finds it, it enables SSH and then deletes the file. SSH can still be turned on or off from the Raspberry Pi Configuration application or raspi-config; this is simply an additional way to turn it on if you can’t easily run either of those applications.

Unfortunately, those instructions don't really work with the rest of the NOOBS instructions.  With a virgin NOOBS installation, the user formats a SD drive using a tool like SDFormatter.  They create a single FAT32 partition according to the instructions for Noobs install (see the link below).

Link to[ "DOCUMENTATION > INSTALLATION > NOOBS"](https://www.raspberrypi.org/documentation/installation/noobs.md)

When you follow the instructions from the www.raspberrypi.org for a Noobs installation there is no /boot/ partition that you can get to.  

Catch 22:
The system works fine if you have a monitor, mouse and keyboard attached.  At initial boot up the user selects an O/S and the system re-partitions the drive to create a fully functional system.  From where I sit, following the instructions to create a SSH enabled boot is simply not possible.  What gives here?   What are the correct instructions for virgin installation "add a `ssh` file to the `/boot/` partition.  

Does this mean the instructions on how to prepare an SD card are wrong?  Or does this mean the instructions on how to add SSH capability at first time boot up are wrong?  ?H

procount, 2016-12-26:
Please see my post on the forum: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=166984&p=1075797#p1075797

zipzit, 2016-12-26:
procount, that's pretty awesome.  

So we can now use this open issue as a request for an upgrade to the  instructions / readme files?  Having the right solution is only part of the conversation.  Why punish others in the future because of mediocre documentation?  

In the release notes included with the /os/Raspbian package (release_notes.txt), where it currently says:

2016-11-25: * SSH disabled by default; can be enabled by creating a file with name "ssh" in boot partition

I'd propose

2016-11-25: * SSH disabled by default; can be enabled by creating a file with name "ssh" in boot partition.  Note: this SSH 'ON' implemention is NOT intended for a NOOBS installation.  NOOBS installs generally require use of mouse/keyboard/monitor.

or suitable words to that effect.

meltwater2000, 2017-03-06:
Surely a ssh option on the recovery.cmdline would solve the problem while maintaining the option to keep security as default.

My reasoning for this, is that beginners are often the ones who really need an easy way to use the Raspberry Pi without being forced to buy HDMI screens.  If headless mode *was* supported out of the box, then a big barrier for new users would be removed.

alecmuffett, 2017-03-06:
s/would solve the problem/would also solve the problem/

Since I have 10x RPi now, anything which lets me solve headless installation with:
* format SD-card
* drag/drop mildly-hacked Noobs files
* eject SD-card
* boot SD-card
* scan network & log in to 8+ RPi at the same time
* use terminal mirroring to upgrade them all.

...is a huge time-saver. :-)

XECDesign, 2017-03-06:
It should be sorted with the next release. 'ssh' and 'wpa_supplicant.conf' files will be copied across, if present.

meltwater2000, 2017-03-06:
That is excellent news.

alecmuffett, 2017-03-06:
@XECDesign - a quick question: how about a generic solution? The above two are fantastic, but a "post-install-run-once.sh" script would be an open hook for people to build upon. 

My credentials in this space: I used to build cluster deployment & orchestration systems on Solaris for Sun Microsystems.

XECDesign, 2017-03-06:
@alecmuffett My thinking is that at that point you should just be creating your own images, but I am not against the idea.

Some thoughts on the matter from a previous discussion: https://github.com/RPi-Distro/pi-gen/issues/47

maxnet, 2017-03-06:
>an extra "post-install-run-once.sh" script would be an open hook for people to build upon.

Can add your post-installation commands to ``os/Raspbian/partition-setup.sh``
Provided you use the full NOOBS .zip that bundles Raspbian, not lite.

meltwater2000, 2017-03-06:
@alecmuffett perhaps @procount's noobsconfig/PINNS is for you!  It has has that and much more, leaving NOOBS to avoid becoming over complicated (I can see both sides, and it has had resistance in the past).

alecmuffett, 2017-03-06:
@XECDesign - I did try creating my own images, but despaired when the docs appeared not to match up with the filesystem layout of unpacked Raspbian.  Also, it made me feel intimidated, that I would have to learn to unpack and repack a distro when all I _actually_ wanted was to configure a hostname on the basis of IP address, this latter already under control because DHCP. 

It was my first real failure and I ended up doing it manually, for lack of an easy way to automate it.

Pretty pictures: https://www.instagram.com/p/BRMOQloDku5/

@maxnet - My thinking was:

* these are, and will remain, headless machines. So I can use Raspbian-Lite, save storage & mem
* because I use Raspbian-Lite, I can use NOOBS-Lite because have no need for Raspbian distro

Therefore, my ideal solution is one which:

* enables network install of R-Lite using N-Lite (yes! install N-lite + wpa_supplicant.conf to SD)
* copies wpa_supplicant.conf to target (not yet, but soon!)
* enables ssh on target (not yet, but soon!)
* copies a script to the target and runs it on first (every?) boot, which I can code to do:
  - add a repo / install some software
  - fetch a script from a repo, run it
  - basically *anything*

Regrettably this latter does not exist, but if it did then all I would need to do in order to reprovision/rebuild my rig is find a way to force it to reinstall. Worst case would be power-cycling it with a keyboard plugged in, hold down shift, and press return twice.

Or - **better** - ask that Noobs kick off the default reinstall process if p1 (not /boot, the other one?) has been mounted and a file created in it called "REINSTALL".

If we had all 5 of those conditions, the rest of a datacentre provisioning solution could be dropped on top of Raspbian quite easily.

procount, 2017-03-06:
NOOBs Is for beginners so there is a limit to how far it will developed. With the recent changes to raspbian's partition_setup.sh, headless setup with wifi and ssh will now be achievable.

If you need further customisation, my noobsconfig hooks into that script to allow additional files to be installed for various "flavours" of the os that you might want.

PINN supports noobsconfig customisations directly without having to hook the script. It also supports ssh so its very useful as an admin/maintenance tool and may support automation in the future. But these are more advanced features that the average beginner won't need.

procount, 2017-03-06:
Look at Meltwaters pikitchen extension to noobsconfig for "run once on 1st boot" functionality.

I think most if not all that you are asking for is already available in some form or other.

alecmuffett, 2017-03-06:
Both those sound interesting - I'll look them up!  

In terms of keeping Noobish things simple, it does seem odd to me to have touch-this-file and edit-that-file and no take-a-step-back "copy this script over and run it", which at some remove could have achieved both of the other things, too. :-) 

But there is excellent precedent for touching magic filenames, eg: how Solaris has dealt with device-tree enumeration for 20+ years: https://docs.oracle.com/cd/E19455-01/805-6331/6j5vgg6aa/index.html

XECDesign, 2017-03-06:
rc.local is something people like to use in the way you describe and beginners struggle with it constantly. A first install script will be simple up to the point a beginner tried to customize it. They don't understand absolute paths, they sprinkle 'sudo' everywhere for good measure, they don't understand why their code doesn't run when they put it after an 'exit 0', they don't understand permissions, users or groups, they don't understand the difference between an interactive shell and a non-interactive shell, they don't understand why their X application doesn't start from the script and they don't understand that they shouldn't paste random things from the internet without understanding it, they don't know how to fix the script if a change breaks it.

My point isn't a condescending "people are too stupid for this feature", and it isn't that running inro problems isn't a good way to learn how things work. I am only saying that scripts aren't easy for people who aren't familiar with the inner workings of a linux distro.

alecmuffett, 2017-03-06:
Yes, I've worked as a university systems admin helpdesk too, you're right, people make mistakes and sometimes it's easier for us as engineers to put the users in a sandbox and stop them playing outside that sandbox.  

But then, isn't education and open extensible platforms one of the goals of Raspberry Pi? I am soon to pick up my soldering iron for the first time in 20 years and have a go at soldering a "ScrollPhat" and frankly I'm a bit worried because if I mess up my PiZeroW then it's a long backlogged wait to get a replacement.  

But that's okay, it's what learning is all about.

XECDesign, 2017-03-06:
I agree with you entirely. The aim isn't to put them in a sandbox and I am not against the feature in principle.

alecmuffett, 2017-03-06:
Imagine what would happen in a world where people learn to stitch their computers together into horizontally-scaled compute fabrics. No more fear of Moore's Law  :-)

niftyhacking, 2018-03-14:
This is good news:  ". 'ssh' and 'wpa_supplicant.conf'"
it might help to also add a file set_hostname 
<dreaming>
Someday a file like UNATENDED-install could contain a template that solves the ssh+wpa+hostname questions to allow an unattended install to the point that the board can be connected to so the remainder of the setup can be done via ssh.  Eventually DHCP+uboot to load a system via  tftp and friends.
https://www.emcraft.com/som/using-dhcp  </dreaming>

lurch, 2018-03-16:
For another first-bootup-configuration solution (although AFAIK it doesn't work with NOOBS) you could look at http://www.pibakery.org/

meltwater2000, 2018-03-16:
I've a modded version of PiBakery which combines PiKitchen, PINN with it so you can (potentially - subject to test) use any standard (including NOOBS) image.

It allows custom scripts and the PiBakery block level config to build a card that would install and run headless configured to your tastes by Judy adding power.

procount, 2018-03-16:
Glad to hear to finally got it working! Would love to play with it.