raspberrypi/noobs #366: Feature request: select OS to boot based on state of GPIO pin(s)

cerebrate, 2017-01-02:
When running a Pi headless with two OSen installed, it would be particularly useful to select the OS to boot each time using a similarly readily-configurable-headless method, such as the state of a GPIO pin.

(This particular suggested method has the advantage that, when the Pi is used with multiple GPIO-using projects, it is possible for the add-on hardware to be configured to automatically choose which OS to boot by setting the startup state of the relevant GPIO pin.)

XECDesign, 2017-01-19:
Thanks for the suggestions. I don't think it's likely that the feature will be added, but we'll keep it in mind.

closed 2017-01-19 by XECDesign

RenardFox, 2017-03-03:
Is that possible to re-consider this request ?
Like cerebrate, I think it would be very useful.

Thanks

XECDesign, 2017-03-03:
There is no objection to the feature itself, but the list of things that would be useful for some people is much larger than the list of things which are helpful for more people and we have time to implement. Sending a PR might be a good starting point.

RenardFox, 2017-03-03:
Thanks for answering, I totally understand.
I'll send a PR.

jharris1993, 2021-03-29:
There is an active solution to this located on PINN's sticky, authored by me - (99% of the credit should go to procount for his patient help!)
[https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&p=1843314#p1843314](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&p=1843314#p1843314)

With the attached script and associated config.txt file, (which you put in PINN's root directory, "RECOVERY"), you can use GPIO pins to select operating systems to boot.

This script assumes you are using one of the last six physical GPIO pins, (35 to 40), to select one of five O/S's.  You can reduce that number or use any other GPIO pin you want so long as it's not being used for anything else.  (See the very important ***WARNING*** within that posting.)

I hope that helps.

lurch, 2021-03-29:
@jharris1993 Nice. If you wanted to make your script more flexible, you could replace the line
```bash
for n in 21 26 20 19 16; do
```
with
```bash
for n in $(grep '^gpio=' /mnt/config.txt | cut -d= -f2); do
```
(but obviously you'd need to replace `/mnt/` with whatever path `config.txt` appears at in PINN - off the top of my head I can't remember what that is).

jharris1993, 2021-03-29:
Actually there's a whole body of work being done on making the entire process more flexible - and that's a great idea! - that is <strike>currently being worked on</strike> currently on hold until I finish my taxes (:wink:)

I want to get my taxes done before the IRS starts taking notice of me, (😨), and get some real project work done, but I am still very interested in getting this totally worked out.

I would say that "99% of the time people won't need anything that generalized" - but we've already seen that "99%" go down the tubes already. . . :laughing: