raspberrypi/noobs #367: Feature Request: Allow Uninstall of OS

cawoodm, 2017-01-04:
This is related to http://raspberrypi.stackexchange.com/questions/59154/
1) Allow uninstall of OS

ghollingworth, 2017-01-04:
It is not possible to uninstall a single OS, this is because you cannot recover the disk space without a significant amount of operating system interaction.

closed 2017-01-04 by ghollingworth

cawoodm, 2017-01-04:
Surely if NOOBS can partition and install an OS it can uninstall it? Having the partitions resized is a nicety and should not defeat the main object. Your average user will not be able to uninstall something they installed otherwise and the issue is compounded by the FORCED 10s delay as soon as you install more than one OS.

ghollingworth, 2017-01-04:
You can just delete it from the os.json file, but this isn't really
uninstalling it, just ignoring it

cawoodm, 2017-01-04:
OK thanks but I'm sure a) this is something lots of people will need and b) there are no simple workarounds. So why not add it to the backlog of wishes?

cawoodm, 2017-01-04:
I meant it when I said this stuff is not simple: Creating a custom OS and deleting everything to remove on OS? As to autoboot.txt: which partition number does one put into autoboot.txt? Is there a guide on how the number is determined, how to mount which partition and create the file? Why not just make it a clickable option in Noobs? Does setting this need to disable the shift option to get to the menu? I understand that is the way NOOBS is *now* but this could be made much better. I appreciate telling me the workaround but I would like these ideas taken up - NOOBS is supposedly for Noobs not Linux masters.

cawoodm, 2017-01-04:
Also there seem to be 2 ways to autoboot: via a `autoboot.txt` and a `noobs.conf`. Not clear which (if not both) are valid.

Traceman, 2017-01-05:
If you need some assistance, please feel free to ask me. Ill help you out if I can. 
Have a good day!
T. Verlin

Date: Wed, 4 Jan 2017 08:44:23 -0800

cawoodm, 2017-01-05:
I ended up [removing the OS](http://raspberrypi.stackexchange.com/questions/59154/re-configure-raspberry-pi-boot-menu/59835#59835). However I now cannot access the NOOBS menu at all, even when holding down Shift while booting.