raspberrypi/noobs #371: Shift on Reboot does not lead to Noobs Menu

cawoodm, 2017-01-05:
When I boot my Raspberyy Pi 3 and hold Shift the entire time I do not get to the NOOBS menu. This was working... until I uninstalled one OS by removing it from installed_os.json. I now have only one OS (Raspian) which I understand bypasses the NOOBS Boot Menu (which is what I wanted) but I would like to be able to get back into NOOBS (like F1 takes me to my BIOS Setup).

lurch, 2017-01-05:
It depends on the particular keyboard, but sometimes holding down Shift doesn't work, and repeatedly tapping Shift works instead.

XECDesign, 2017-11-03:
Answered

closed 2017-11-03 by XECDesign

nuklearwintr, 2017-11-08:
Tapping the shift key worked for me, but holding it did not.  I have a wireless usb keyboard.

vegerot, 2019-10-15:
I've tried rapidly tapping and holding shift, and neither work

lurch, 2019-10-16:
@vegerot What sort of keyboard is it? If you can, try using a wired USB keyboard.

vegerot, 2019-10-16:
I am using a wired keyboard.  However, if it is relevant, it's an RGB mechanical keyboard.   What I've noticed is that the lights on the keyboard don't turn on until maybe 20 seconds after I turn on the machine, or a few seconds before the system finishes booting.

BTW after doing more research yesterday I thought it was no longer possible to boot into NOOBS by using Shift, and that that was disabled after the first OS is installed.

lurch, 2019-10-16:
Sounds like the RGB lights are drawing too much current, and the keyboard isn't ready in time when you're trying to press Shift?

vegerot, 2019-10-16:
That's what I'm thinking.  Or the software inside the keyboard (that controls the custom patterns and themes) takes a certain amount of time to load, and that's longer than the window to enter NOOBS

lurch, 2019-10-16:
Perhaps a workaround might be to use a powered USB hub between the Pi and your keyboard, and then power on the hub first so that your keyboard has enough time to 'get ready', and then power on the Pi? :man_shrugging: