raspberrypi/noobs #372: Custom Boot Splash

orgertot, 2017-01-07:
Good morning,
Apparently nobody ever wanted to change the boot splash. All tutorials for other os' do not work. 

I figured out that you can easily change the boot splash in the folder /usr/shared/plymouth/themes/pix/

My problem now is:
-between the boot splash and the desktop the monitor becomes black for like 3-5sec. How do I remove that or extend the boot splash to be shown? 
-how do I remove:
--the colorful gpu Test in the beginning
--the recovery screen

Hopefully we will find together a answer so we could help others who want to archive the same! 

Thank you very much

ghollingworth, 2017-01-14:
Between the boot splash and the Raspbian desktop the Pi is actually reset, this is why you get the black screen, it is not possible to continue to display on the screen through a reset.

You can remove the GPU test screen using disable_splash=1 in config.txt

What do you mean remove the recovery screen?  How would you get to NOOBS if you removed that?

Gordon

lurch, 2017-01-15:
It's possible to bypass the recovery screen by using `autoboot.txt` - https://github.com/raspberrypi/noobs#how-to-bypass-the-recovery-splashscreen-and-boot-directly-into-a-fixed-partition

XECDesign, 2017-11-03:
No response

closed 2017-11-03 by XECDesign