raspberrypi/noobs #374: Language native name not showed right

bovirus, 2017-01-17:
In file

\recovery\languagedialog.cpp

There is this code to read and show language name in native language

        QString langcode = langfile.mid(12);
        langcode.chop(3);
        QLocale loc(langcode);
        /* Display languagename in English, e.g. German, French */
        /* QString languagename = QLocale::languageToString(loc.language()); */
        /* should Display languagename in native language, e.g. Deutsch, Français  */
        QString languagename = loc.nativeLanguageName();
        QString iconfilename = ":/icons/"+langcode+".png";

I have a problem with 

Spanish (file language_es.ts) - code=es

Language name showed "Espanol de espana". Is it right?

procount, 2017-01-20:
QT only seems to support 2 character country codes from ISO 639-1.  How should we support longer country codes and what format should they use?

Chinese (Taiwan) is already indicated as 'zh_TW', so is this the correct format?

Asturian could be: 'es_XA', 'es-XA' or 'ast'. But using 'es_XA' results in a native language name of "espagnol de Espana", which is the same as Spanish and probably not correct. Is that because it just uses the first 2 letters of the country code? And does that mean 'zh_TW' is simply being displayed as 'Chinese'?

Basque could be 'eu' or use it's subdivisions of 'eu_ES' or 'eu_FR'.

Probably the best way is to detect the longer country codes and provide a separate set of language names for them in a separate translatable language file that translators can edit to augment QT, but what is the correct language code format to use for these?

bovirus, 2017-01-20:
@procount 
I believ that is more simple to add a string value (ex. NativeLanguagename) in .TS language file and use tio show the native language name.

Bsque = eu
Asturian = es_XA

XECDesign, 2017-03-14:
Fixed by https://github.com/raspberrypi/noobs/commit/5136f34c9aa0a2550b89eb17bc07aaad1d1fbf3b

closed 2017-03-14 by XECDesign