raspberrypi/noobs #376: static ip

max099521, 2017-01-19:
Dear all 
i have a question about static ip
If i want to use a static ip 
i should change then /etc/network/interfaces ?
i try to change the interfaces like .
..................................................................
auto lo eth0
iface lo inet lookback
iface eth0 inet static
192.168.0.1
255.255.255.0
192.168.0.250

..........................................
and reboot it 
but it look like doesn't walk .
so .  
what's mistake i make  ?
Isn't any one can teach me ?

procount, 2017-01-19:
Do you really want to fix the IP address in the NOOBS os installer program itself, or do you mean to set it in the Raspbian OS that you installed using NOOBS?

max099521, 2017-01-19:
.i have use a noobs install os .
and i want to backup my os  
so ..
i want the static ip can be backup to the os
and 
i stall want can change it  if i already install the os with NOOBS

procount, 2017-01-19:
So this is not really a NOOBS related issue, but a question about how to set up a static IP on an OS that was installed by NOOBS. But you didn't state what operating system you installed. I can only assume it is Raspbian. there are plenty of posts on the forum about setting up a static IP address in Raspbian, but the method you outline above is from the old way of setting a static IP with Raspbian Wheezy. In the latest Raspbian Jessie, you should be using dhcpcd.conf in preference. Please see the forum for help on how to do that. Or you could just configure your dhcp service on your router to allocate a static IP address.

But I really don't understand why you need to set a static IP in order to backup your OS.
If you do have the latest Raspbian installed it includes the piclone SD Card Copier which can backup your whole card to another one in a USB card reader.
You could use Win32DiskImager on Windows, or DD on Linux. To save space you could just TAR each partition into an archive file, then comrpess it with gzip or xz (like NOOBS stores its compressed OSes)

XECDesign, 2017-01-19:
If this is a stock Raspbian question, restore the original interfaces file and see /etc/dhcpcd.conf.

closed 2017-01-19 by XECDesign

max099521, 2017-01-20:
thanks all ..i think i got the point .
i was use the Jessie so the static ip  using dhcpcd.conf 
thanks all
how to set dhcpcd.conf  subnetmask.

lurch, 2017-01-20:
> how to set dhcpcd.conf subnetmask.

That's a question for the [forums](http://www.raspberrypi.org/forums/) :)