raspberrypi/noobs #378: Changing recovery shortcut key [shift] to a custom buton/key

eddywebs, 2017-01-23:
Hi, we are creating a gaming console operating system based off noobs, would it be possible to change the default recover key [shift] to a custom key or possibly a keypress from a usb controller ?

XECDesign, 2017-01-23:
Do you mean in the official release or in the source code? I would recommend forking NOOBS and making the change here (https://github.com/raspberrypi/noobs/blob/master/recovery/main.cpp#L187) for your build.

eddywebs, 2017-01-24:
>

closed 2017-01-24 by eddywebs

lurch, 2017-01-24:
It probably depends whether support for your USB device has been [configured](https://github.com/raspberrypi/noobs/blob/master/buildroot/kernelconfig-recovery.armv6) in the NOOBS kernel.

maxnet, 2017-01-24:
- make menuconfig, switch Qt embedded keyboard driver to linuxinput (evdev)
- remove existing keymaps, provide your own, and make sure that one is always used in the init script.

https://github.com/raspberrypi/noobs/tree/master/buildroot/package/recovery
http://doc.qt.io/qt-4.8/qt-embedded-charinput.html

eddywebs, 2017-01-26:
@maxnet So turns out the make menuconfig does not have any Qt keybord driver options,  can anybody confirm the location ?

maxnet, 2017-01-26:
>@maxnet So turns out the make menuconfig does not have any Qt keybord driver options, can anybody 
>confirm the location ?

Sorry, but I'm not going to provide a tutorial that detailed.
Can search for options in kconfig by pressing /