raspberrypi/noobs #379: bsdtar problem

procount, 2017-01-27:
@maxnet - I see you now included bsdtar instead of tar, like i did in PINN.
I had a request to include Voidlinux in PINN but when I tried to decompress it I got the following error:

bsdtar: Pathname can't be converted from UTF-8 to current locale. 

It seems to be a locale issue and probably only a warning, although it is treated as an error so the installation stops. The following URL might be useful for background info: https://github.com/docker/docker/issues/355

I tried preceding the cmdline with LANG=en_US.utf-8 in the shell, but it still came up with an error.
Any ideas how to resolve this? tar still works fine on the voidlinux file.

maxnet, 2017-01-27:
Try preciding with LOCALE=POSIX when CREATING the archive. That gives warning when creating, but not when extracting. And will store the filenames in binary format in the tar.

Adding locale support to NOOBS is problematic.
uclibc C library version we currently have does not have the locale option enabled, and did not compile on first try last time I tried.
Upgrading uclibc or changing to glibc can also increase footprint. And could potentially cause binary compatibility issues with the Windows installation software components.

procount, 2017-01-27:
Voidlinux is already provided in .tar.xz format so I might have to ask the maintainer to change how they compress it.

Can I tell what method they used to compress it i.e. tar, gnutar or bsdtar (posix or otherwise?)

Failing that, I wonder if we need to identify whether an os needs bsdtar or tar?

anshikam, 2017-05-01:
I am getting the same error when installing Retropie using NOOBS.
bsdtar: Pathname can't be converted from UTF-8 to current locale.
bsdtar: Linkname can't be converted from UTF-8 to current locale.

I tried creating the tar file and archiving through the commands
$ sudo tar  --format=posix -cvpf ~/Desktop/retropie.tar ./*
$ LANG=POSIX xz -9 -e -v retropie.tar

My Retropie image which I am trying to tar is more than 2GB in size.
Any resolutions to this yet?

maxnet, 2017-05-01:
>I tried creating the tar file and archiving through the commands
>$ sudo tar --format=posix -cvpf ~/Desktop/retropie.tar ./*
>$ LANG=POSIX xz -9 -e -v retropie.tar

LANG=POSIX belongs to the first line.

closed 2017-11-03 by XECDesign

xilopaint, 2017-11-05:
I got the same error when trying to install [pi64](https://github.com/bamarni) with NOOBS. In my case I tried this:

```
$ LANG=POSIX sudo tar -cvpf ~/Desktop/pi64.tar ./*
$ xz -9 -e -v pi64.tar
```

And the same for boot.tar.

I used [this guide](https://github.com/retropie/retropie-setup/wiki/convert-retropie-sd-card-image-to-noobs-image) for RetroPie.

Thoughts?

XECDesign, 2017-11-06:
By default, sudo doesn't preserve all of the environment variables. Put `LANG=POSIX` after `sudo`, not before it.

xilopaint, 2017-11-06:
I managed to skip the error using `bsdtar` instead `tar`:

`LANG=POSIX sudo bsdtar --numeric-owner --format gnutar -cpf ~/Desktop/pi64.tar .`
(now I know `LANG=POSIX` made no difference)

At the moment I'm struggling with another issue:

[image]

So close... 😞

procount, 2017-11-06:
@xilopaint - When you convert an OS for use with NOOBS, you need to remove such resizing options because they do not apply: NOOBS has already expanded the partition to the required size before the OS is installed. So the problem is not that it cannot expand the filesystem, but that it is trying to expand it at all. Find the boot script that causes this and delete it.

xilopaint, 2017-11-20:
It worked. Thanks @procount!