raspberrypi/noobs #380: multiImageWriteThread error

procount, 2017-01-27:
There is a debug message "QObject: Cannot create children for a parent that is in a different thread" that occurs ~3 times when writing an image.
I think it might be due to the new MultiImageWriteThread::addImage() use of new osInfo() which is a shame, How are you supposed to use QObjects to pass data between threads?
This might be a source of unexpected behaviour.

maxnet, 2017-01-28:
The parent/child thing is just for memory management.
If you ``delete`` an object, the ``~QObject`` destructor will take care of deleting all registered children automatically, so lazy developers do not have to call ``delete`` on those objects manually.

So if it fails to register a child, the child will live beyond his parent, and you have a memory leak.
That's a bug alright, but I would not worry to much about it. 
Will not get unexpected behavior beyond that.

The ``MultiImageWriteThread`` object belongs to the main thread, because it was created there, even though it symbolizes another thread.
``addImage()`` is called from the main thread as well, so that should be fine.
Problem is more likely the ``new PartitionInfo`` calls elsewhere in the file that do run in the new thread, but designate ``this`` as parent. 
Shouldn't use those relationships in that case, but ``delete`` those objects manually instead when no longer necessary.

procount, 2017-01-28:
Yes, I concluded it was the osInfo - partitionInfo relationship that caused the issue. I wondered if the warning was also to warn against accessing the structure simultaneously from different threads, thus requiring a mutex? Of course that's fortunately not an issue here.

Is it safe to manually change the parent of a QObject so that it will get destructed in the new thread? 
Will copying a QObject parameter from another thread result in the new QObject having the correct parent? 

I ask because I'd like to do some groundwork harmonising some of the inconsistencies and I'm trying to avoid these issues.

maxnet, 2017-01-28:
>Is it safe to manually change the parent of a QObject so that it will get destructed in the new thread?

Should set the right parent when creating the QObject, or do not set a parent at all and destruct the thing manually.

>Will copying a QObject parameter from another thread result in the new QObject having the correct parent?

If you copy a pointer to an object, nothing will change to said object.
QObject's aren't meant to be cloned.
Unlike say a QString that does have a copy constructor.

XECDesign, 2017-11-03:
No activity

closed 2017-11-03 by XECDesign