raspberrypi/noobs #382: add rtandroid

enboig, 2017-02-02:
Is there a way to add https://git.embedded.rwth-aachen.de/rtandroid/downloads/raspberry-pi/ to noobs? if not in the "official" version, a tutorial to modify the os.json (or other files needed).

Thanks

lurch, 2017-02-02:
This sounds like one for @procount ;-)

procount, 2017-02-02:
It would help if the image was split into individual partitions, preferably changed to a .tar.xz format and an icon and slides were made available.

The README.md file has some details on how to do this to create a custom os and I think there is an article in the wiki too.

procount, 2017-02-03:
It would still help if an icon and slides were made available for this installation.
The image processing and installation looks like a fun challenge to automate :)

procount, 2017-02-04:
I'm not sure if this is going to be possible without a special build of rtandroid.
Although I can get it to boot to the "A N D R O I D . . ." screen, it won't go any further.
I coudn't find any config files to indicate the locations of the partitions (like the usual cmdline.txt and fstab that need adjusting).
I've posted some questions to see if its possible so we will see.......

XECDesign, 2017-03-03:
It would be up to rtandroid to make rtandroid installable within NOOBS. In my brief bit of testing, it didn't seem quite ready yet.

closed 2017-03-03 by XECDesign

procount, 2017-03-16:
I finally managed to get RTAndroid to load manually within a NOOBS like environment, so I hope to have it available in PINN soon once I've added the support files for it.

procount, 2017-03-17:
RTAndroid 7.1.1 has been added to PINN.
RPF and Waveshare touchscreen support needs to be added manually for now.

In order to work with NOOBS, cpio will need to be added from Busybox, and possibly PR #381.
This is a one-off conversion of the latest release, but it looks like they are willing to add support for a NOOBS compatible installation into their build process.