raspberrypi/noobs #385: Change select os  wait time

max099521, 2017-02-10:
Dear all
i have some question 
1. If i have 2 OS .Can I change the select OS waiting time ? (default is 10 Sec)
2 .If i use SSH TO LOGIN . Can I Change Boot to another OS?

Thanks

procount, 2017-02-10:
1. Not with NOOBS. This feature is included in PINN by adding 'bootmenutimeout=X' to the recovery.cmdline. (Could be added to NOOBS in the future)

2. Do you mean SSH into Raspbian or NOOBS? 
NOOBS does not support SSH. You can add 'vncinstall' to recovery.cmdline to use VNC with NOOBS in order to see the NOOBS screen, so you can remotely change the OS to boot in that way on boot up. When using VNC it is often a good idea to add 'forcetrigger' as well, otherwise you might miss the splash screen and will be unable to go into the main menu. Using 'vncinstall' has the disadvantage that any local display remains blank whilst NOOBS is running on the remote VNC screen. PINN includes 'vncshare' so that the screen can be displayed both locally and remotely at the same time. 

PINN includes SSH. Whether you log in via SSH to PINN or another OS that NOOBS/PINN has installed, the way to change the default OS to boot is to change the NOOBS setting file to direct NOOBS to boot a specific OS if there is no user interaction. Alternatively, you can create an Autoboot.txt file to bypass NOOBS and boot directly to one of your installed OSes. PINN also includes a 'sticky default' boot option that bypasses the OS boot selection dialog to boot a specific OS more quickly.

(Please see the NOOBS README.md or PINN README_PINN.MD documentation files which include a lot of these details and more. If something is missing, please let us know).

max099521, 2017-02-10:
thanks procount
I'm backup my OS..
I'll Try it later . Thanks your answer.
Have a day

lurch, 2017-02-10:
Duplicate of #168

closed 2017-02-10 by lurch