raspberrypi/noobs #386: How do I enable channel 13 wifi networks ?

maicod, 2017-02-26:
my wifi-network uses channel 13 because its the most unused channel here. Can I edit a configuration file in NOOBS so it sees my network ?

XECDesign, 2017-02-27:
It's not pretty, but it looks like the only way right now is to drop a wpa_supplicant.conf file on the settings partition (which wouldn't exist right after copying the NOOBS files).

XECDesign, 2017-02-27:
Actually, my mistake, it can be dropped onto the main (first) partition.
https://github.com/raspberrypi/noobs/blob/d709621c1f80700b4ea058a8910a24d98b47be4e/recovery/mainwindow.cpp#L942

maicod, 2017-02-27:
I acquired the wpa_supplicant.conf file from a Raspbian Jessie filesystem where the wpa_supplicant.conf was already filled in with my network settings cause I had booted/configured that raspbian before with my Pi 3B. I copied it to the root of the fat partition where the other Noobs files reside and booted, but it still doesn't login into my wifi or show the SSID in the network list.

XECDesign, 2017-02-27:
It copies the file only if it does not already exist on the settings partition. You might have to copy over it, remove it or start from scratch.

maicod, 2017-02-27:
It still doesn't login. It copied the wpa_supplicant.conf file to the settings partition succesfully and afaik it is correct. It looks like:
`
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=NL

network={
	ssid="myssid"
	psk="mypassword"
	key_mgmt=WPA-PSK
}
`

XECDesign, 2017-02-27:
Alright, I'll give this a go tomorrow and get back to you.

XECDesign, 2017-02-28:
I haven't had any problems with this procedure:

1. Create a new partition table on the card.
2. Create and format a FAT32 partition.
3. Copy NOOBS Lite files and wpa_supplicant.conf onto the partition.
4. umount/eject and boot the pi with the card.

NOOBS boots and connects to wifi automatically, showing that wpa_supplicant.conf is being used.

procount, 2017-02-28:
I think the general procedure is fine. More likely to be a problem of using Channel 13. Is that channel enabled by the NL country code? Is there something else that needs to be done to enable channel 13?

XECDesign, 2017-02-28:
At least in Raspbian, it is only a matter of changing the country in wpa_supplicant, but NOOBS may very well behave differently.

Ruffio, 2017-02-28:
The setting in wpa_supplicant can be 'overridden' if the router is configured for something else. In my case, I set the wpa_supplicant to DK, but apparently the router is configured for DE, so I can see in the dmesg that is switches: https://github.com/raspberrypi/firmware/issues/630

maicod, 2017-02-28:
@XECDesign Thanks for trying it out. Did you try it out when having set your wifi network to channel 13 ? I retraced your steps but still don't see my network. btw Its NOT a hidden network. I now found out that the Ubuntu Mate for Raspi has the same problem. Raspbian has not. I suspect its a kernel or module thing that physically blocks reading channels 12,13 so setting it up manually with wpa_supplicant.conf won't help, it still can't/is not  allowed to read channel 12 and 13. I found a way around the problem by using wired connection but thats fine for NOOBS as its only an installer but for Ubuntu Mate its not ( it needs to be connected to wired ethernet all the time then)

maicod, 2017-02-28:
@procount I tried it with countrycode GB also to no avail

procount, 2017-03-01:
There are no CRDA messages in NOOBS dmesg output, so my uneducated guess is that the driver is just using the default world (subset) of channels that are legal everywhere. Would this required BR2_PACKAGE_CRDA=y ?

XECDesign, 2017-03-01:
@procount Yes, I believe you're right about it not working without having CRDA.

maicod, 2017-03-02:
@XECDEsign @procount so thats it, its a kernel build option then. Thats apparently the reason Raspbian does see channel 13 and Ubuntu Mate and NOOBS do not .

XECDesign, 2017-03-02:
A buildroot option to add the CRDA package. Will have to test it properly to confirm that that's all it is. Thanks @procount

XECDesign, 2017-03-03:
Adding CRDA alone does not make it work.

procount, 2017-03-03:
I did a bit of research.... I think CRDA is designed to be used as a helper function for udev, but that isn't used in NOOBS is it? May need another way of triggering it, or building db.txt directly into the wifi stack? <shrug>

XECDesign, 2017-03-03:
That worked

procount, 2017-03-03:
Yay!    \o/

XECDesign, 2017-03-03:
New NOOBS being uploaded now. After providing my own wpa_supplicant, I can see the regulatory domain switching in dmesg.

maicod, 2017-03-03:
@XECDesign wonderful ! I'll try it out.

XECDesign, 2017-03-06:
Any luck?

closed 2017-03-06 by XECDesign