raspberrypi/noobs #387: Formatting Instructions

ronaldpetty, 2017-02-26:
Hello,

I am trying to start from absolute 0 with a micro SD card (16 gb).  I have so far been unsuccessful in formatting the card on Mac Sierra.  I can use the SD Formatter tool, but that is cheating.  I am trying to use diskutil only.  I am unable to find complete examples online, or I have not been able to understand them.  Here is what I have done so far:

```
diskutil unmountDisk /dev/disk2
diskutil partitionDisk /dev/disk2 GPT DOS_FAT_32 NOOBS 0b
cd /Volumes/NOOBS/
cp ~/Downloads/NOOBS_v2_1_0.zip .
unzip NOOBS_v2_1_0.zip
rm NOOBS_v2_1_0.zip
cd
diskutil unmountDisk /dev/disk2
diskutil eject /dev/disk2
```

Place into Pi, power on, nothing happens, no blinking, no booting.  Can someone point me in the right direction?

XECDesign, 2017-02-26:
I would recommend asking on the forum, as this repo is for reporting issues and I don't know how many Mac users keep an eye on the reports here.

closed 2017-02-26 by XECDesign

ronaldpetty, 2017-02-26:
Thanks @XECDesign .  If I do figure it out, would there be interest in a PR for SD formatting docs?

If anyone does land here, the part that was not working was GPT option.  I read many docs stating that was the way to go, but if I use MBR it works.

XECDesign, 2017-02-27:
That might be an issue for the documentation team. https://github.com/raspberrypi/documentation/issues