raspberrypi/noobs #390: autoboot.txt not working in NOOBS Lite 2.3 (Regression from 2.1)

matthuisman, 2017-03-08:
Steps to Reproduce:

- Download [NOOBS Lite 2.3](https://downloads.raspberrypi.org/NOOBS_lite/images/NOOBS_lite-2017-03-03/NOOBS_lite_v2_3.zip)
- Write to fresh FAT32 formatted SD Card
- Put into Pi and boot
- Install LibreELEC and Lakka
- Remove SD Card.
- Create **autoboot.txt** file on root of SD card with contents **boot_partition=6**
- Put into Pi and boot

Expected Result: NOOBS boots into either LibreELEC or Lakka
Actual Result: NOOBS boots into system selection screen

Have tested [NOOBS Lite 2.2](https://downloads.raspberrypi.org/NOOBS_lite/images/NOOBS_lite-2017-02-27/NOOBS_lite_v2_2.zip) and has the same issue.

[NOOBS Lite 2.1](https://downloads.raspberrypi.org/NOOBS_lite/images/NOOBS_lite-2016-11-29/NOOBS_lite_v2_1.zip) works as expected.

Raspberry Pi 3 - Model B
Samsung EVO 16GB SD

matthuisman, 2017-03-08:
You can just install 1 system for a quicker test.
But if only 1 system, NOOBS will auto boot into that.

But you can still tell the autoboot is not working due to the rainbow screen and a quick flash of the NOOBs selection window.

XECDesign, 2017-03-08:
Interesting. I am not seeing any reference to autoboot.txt or boot_partition in code.
@lurch Any idea how this ever worked?

XECDesign, 2017-03-08:
Found that it's a firmware feature. Looking into it.

lurch, 2017-03-08:
@XECDesign Yeah, when `autoboot.txt` is present it's read by the firmware, and NOOBS (on partition 1) gets skipped entirely.

lurch, 2017-03-08:
P.S. This isn't the first time that a newer firmware update has (inadvertently) stopped this feature working ;-)  See https://github.com/raspberrypi/noobs/issues/195 - hopefully the fix will be just as easy this time round.

procount, 2017-03-08:
Yes. It's a firmware feature. I confirm it's broken in PINN too. 
Here is another interesting side-effect....
When I set autoboot.txt, additionally with the NOOBS forcetrigger option, PINN is still executed and it goes to the OS selection window, but none of the OSes present on the USB stick are displayed. Yet the USB stick is still mounted and accessible. Not able to test downloadable OSes, but I'm not going to debug further cos it clearly is a firmware issue.

XECDesign, 2017-03-08:
Forwarded to @ghollingworth

lurch, 2017-03-08:
@procount Sounds like your issue is actually something else - AFAICT this particular firmware bug _only_ means `autoboot.txt` isn't working, it shouldn't have anything to do with USB drives. Maybe by setting the `forcetrigger` option, the PINN interface is being displayed much quicker than it normally is, and so the USB drive hasn't _actually_ got mounted by the time PINN first tries scanning it? (but by the time you login to poke around, the USB drive has had time to mount?)

procount, 2017-03-08:
Well, I don't doubt there is another issue there (which will be solved when I re-architect...), but this side-effect is caused by the autoboot.txt issue (probably it is altering the boot up timing). Removing the autoboot.txt file allows it to work normally and I normally use forcetrigger anyway cos I run headless 80% of the time. and it was fine before. (It's just an interesting side-effect in case it helps diagnosis)

matthuisman, 2017-03-08:
@procount 
How does USB installed systems work with autoboot.txt?
I thought autoboot could only contain the boot partition number for a mmcblk partition?

procount, 2017-03-08:
@matthuisman - don't misunderstand this PINN feature - It uses the USB stick as a SOURCE of compressed OS files to be installed from. It does NOT install an OS to a USB stick. 

The autoboot.txt and reboot features are a feature of the firmware, so they work equally well with NOOBS and PINN. 

I expect your KODI plug-in is already compatible with PINN. Why don;t you try it......

matthuisman, 2017-03-08:
@procount 
Thanks for clarifying.

Also, a quick feature suggestion for PINN (and hopefully NOOBS)

Optionally allow user to edit partition sizes after selecting OS's to install.
First as normal use each OS's "partititons.json" file to generate an initial layout.
Then if the user presses "P" or something like that, they can edit this layout (just set sizes - not move partitions).

It would only allow adjusting the partitions that have "want_maximised" and would not allow setting smaller than the "partition_size_nominal" size.

This would allow users using say LibreELEC + Recalbox to give a lot more space to Recalbox on larger SD cards.

In code, this could set wantMaximised to false and adjust partitionSizeNominal to the value set by user.

Thoughts?

lurch, 2017-03-09:
> Optionally allow user to edit partition sizes after selecting OS's to install.

See #139 for an explanation of why this feature is unlikely to be added.

matthuisman, 2017-03-10:
Thanks.
I have posted my idea over there.

Also, is it known that NOOBS seems not able to get an os_list_v3.json from an HTTPS url?
However, it can retrieve the other files just fine from HTTPS (os.json, partitions etc)

maxnet, 2017-03-10:
>Also, is it known that NOOBS seems not able to get an os_list_v3.json from an HTTPS url?

https needs correct date and time settings, since certificates do not only have an expiration date, but also a "not valid before" date.

The Pi does not have a RTC, and will think it is 1970 at boot, so https will not work then.
After it obtained the date/time from the HTTP server headers it receives when downloading the os_list  subsequent requests do can use https.

procount, 2017-03-10:
@maxnet - I resolved this in PINN. Well, enough for my uses anyway.....
Move the time setting from downloadListComplete() to downloadListRedirectCheck(). 
Specify a http: url instead and rely on the webserver redirecting to a https: url.

It's not a solution to making the download secure, but it is a way of connecting to a URL that only provides a https: feed.

XECDesign, 2017-04-10:
This should be fixed with https://github.com/raspberrypi/firmware/commit/b038854bac01ac2cb772de19744a7bb956a3b170. The fix didn't quite make it into NOOBS 2.4, but should be picked up in 2.5.

procount, 2017-04-10:
Thanks for the info.
BTW, anything else new coming in 2.4 or have you just to bump the version? will it be soon?

XECDesign, 2017-04-10:
No, 2.4 is mostly just USB support. No new commits coming. If all goes well, it should be live today.

procount, 2017-04-10:
OK. Great.
Previous rpi-firmware was modified to use the stable branch instead of master.
Is it safe to go back to the master branch to pull in raspberrypi/firmware@b038854 ?

XECDesign, 2017-04-10:
The difference between stable and master should only be the kernel, so it shouldn't matter.

XECDesign, 2017-04-10:
Be careful if you update the kernel to 4.9, by the way, NOOBS doesn't currently work with the new reset controller driver.

procount, 2017-04-10:
Thanks ;-) I can wait. (I prefer to build off your stable builds anyway)

maxnet, 2017-04-10:
>Be careful if you update the kernel to 4.9, by the way, NOOBS doesn't currently work with the new reset controller driver.

And be careful if you want to use master firmware with 4.4.
The bundled DTB files meant for 4.9 are known to hang 4.4

You would need to compile the DTS files from 4.4 kernel source.
Buildroot can do that for the main ones, but does not support the compiling overlays, as that is something non-standard (the concept of overlays doesn't exist in upstream Linux).

procount, 2017-04-10:
OK. Let me get this straight (cos it's rapidly getting confusing...)
If I take NOOBS 2.4 as-is, and update rpi-firmware to b038854, it won't work, because this commit also brings in the dtbs and dtbos from the 4.9 kernel, which are not compatible with NOOBS' 4.4 kernel and cause it to hang? So I would have to use the dtbs and dtbos from the 4.4 kernel (that are already included in NOOBS 2.4) and compiling them from source could be an tricky.
So howabout I just import the start*.elf files from b038854 and leave everything else the same - would that work? Or does this bring in another compatibility issue (I'm beginning to see why you didn't include it in 2.4!)

XECDesign, 2017-04-10:
22de0bb68d34fd210ba9d086c6a1fc5e90f0bfbb should be safe

procount, 2017-04-10:
Thanks guys.
I'll try it out, but not tonight.

XECDesign, 2017-04-11:
Thanks for the info. Does what you're booting into have up to date firmware?

procount, 2017-04-11:
Good question. Should that matter? 

I tried it with  OSMC (since I was testing the partition_setup.sh script the other night,) and Void linux (because it is small and doesn't take long to install) and both of them booted fine when selected from NOOBS menu. Both of them boot fine with autoboot.txt and when the noobs/bootcode.bin is from 2.1

Which part of the firmware in the OS do you mean - bootcode.bin, start.elf? I'm sure OSMC was the latest from the internet (as I was testing the setup script). Void is from Feb 10th? so before the PiZeroW, so not the latest. 

I can try the latest Raspbian-Lite to see if it makes a difference, but it'll have to wait til later.

Where does autoboot.txt fit into or alter the boot sequence?
Is there some interaction between the files on NOOBS and those in the OS when it boots, leading to an incompatibility? I thought they were totally separate,,,

XECDesign, 2017-04-11:
bootcode.bin reads autoboot.txt and can then determine where to load start.elf from. The problem was that start.elf was not aware of where to load everything else from, because that information was discarded by bootcode.bin. What would happen is that you could have some of the files loaded from the correct partition and the rest from an incorrect partition.

The fix was to store the partition number to boot from in a spare register, but both bootcode.bin and the loaded start.elf have to be aware of it.

procount, 2017-04-11:
But that isn't anything new is it? I mean, AIUI from the beginning of noobs multi-boot, the partition number was stored in some spare non-volatile register bits that survived a warm reboot in order to pick up start.elf from the right partition. So all the OSes that are available though noobs are new enough to support that aren't they? Or are you talking about a more recent change, like the register used to pass the partition number has changed, or more info has had to be stored for USB multi-booting?

XECDesign, 2017-04-11:
I am being a little vague because I don't remember the specifics and don't want to say something that's completely wrong. At some point in the past, the value must've been written into the reset register. Something about the early USB and network boot commits partially broke that mechanism. It read the value from autoboot.txt, but didn't write it into rsts. That register contains a value which can be useful (it can tell you that you've had a watchdog reset, for example). By using a different register, that value is preserved in more cases.

procount, 2017-04-11:
OK, that's fine. It gives me an idea and some understanding of why newer firmware is essential, anyway and what to look for in my tests. Thanks.

procount, 2017-04-11:
Raspbian full & Lite both seem to work with Autoboot with noobs 2.4. Looks like some OSes might need updating.

matthuisman, 2017-04-23:
So is autoboot fixed in 2.4?
I'm trying your latest PINN @procount and autboot still seems no go.

Lakka stuck @ colour pattern - this is pretty bad :(
LibreELEC at least still boots into PINN (but not autoboot)

procount, 2017-04-23:
Yes and No.
As I understand it from the conversation above, the behaviour of how to boot into a particular partition changed a while ago for better stability. 
In the boot process, bootcode.bin boots first, works out which partition to boot into (from autoboot.txt or from NOOBS) and then loads start.elf from that partition. Start.elf then loads the next part, but it needs to know which partition to load it from. bootcode.bin forgot to pass this information along, but now this mechanism is more reliable. However it means that bootcode.bin {from NOOBS) and start.elf (from your OS) need to be compatible.
The problem we have now is that NOOBS/PINN are using a newer bootocde.bin and I suspect some of the OSes still use an older start.elf, so it doesn't know where to load the OS from.
**So the proper fix is to upgrade the OS to the latest RPI firmware.**
Where this is not possible, I found I could downgrade bootcode.bin in NOOBS (say from NOOBS v2.1) and it may work with some of the older OSes. Your mileage may vary!
(This is just my understanding. I'm sure @XECDesign will correct me if I got anything majorly wrong :) )

matthuisman, 2017-04-23:
Are all start.elf the same?
Could my add-on update the start.elf on older systems?

procount, 2017-04-23:
Pass... 
To do it properly, you would probably have to upgrade bootcode.bin, start*.elf, *.dtb, *.dtbo and _maybe_ the kernel and its modules too (?). The GPU firmware is all closed source, so the relationship between all of these elements is best explained by people like XECdesign. It may even change between different releases.
(Or just try it and see....)

matthuisman, 2017-04-23:
No worries.
Sounds a bit too much to try to modify.
I've gone for the "warning" approach until enough systems are compatible.

Do we know the minimum firmware version for OS's that is required to work with autoboot?

XECDesign, 2017-04-24:
This isn't fixed in NOOBS 2.4.

The initial boot partition (NOOBS, PINN or whatever else), needs to have an up to date bootcode.bin  and the boot partition you're booting into needs to have an up to date start.elf.

By 'up to date' I mean firmware that has the 'firmware: rebuild with Avoid PM_SPARE bit 23' commit:
https://github.com/raspberrypi/firmware/commit/22de0bb68d34fd210ba9d086c6a1fc5e90f0bfbb
https://github.com/raspberrypi/firmware/commit/b038854bac01ac2cb772de19744a7bb956a3b170
https://github.com/Hexxeh/rpi-firmware/commit/3ff94f1fc459f88d3e2530542fb609643a7bd1a6

procount, 2017-04-24:
(I sneaked raspberrypi/firmware@22de0bb into PINN 2.4)

matthuisman, 2017-04-26:
@procount

Any idea where "TLXOS" is coming from in the PINN OS list?
It's not in my .json file and I can't see it in os_list_v3.json either...

'TLXOS - 60 day trial of ThinLinX's Debian-based thin client'

procount, 2017-04-26:
Do you have it on your SD card in /os, or on a USB stick in /os ?

matthuisman, 2017-04-26:
Nope. 
Double checked. 
OS is empty. 
No USB drives in.
It must be coming from Raspberry Pi...

Reading their Tweets - it sounds like it's been added.
https://twitter.com/thinlinx/status/838896528106258433

Just confused as it's not in 
https://downloads.raspberrypi.org/os_list_v3.json

lurch, 2017-04-26:
> Just confused where it's coming from as it's not in
https://downloads.raspberrypi.org/os_list_v3.json

I see this at the bottom of that file:
```json
        {
            "description": "60-day trial of ThinLinX's Debian-based thin client for Raspberry Pi 2 & 3",
            "icon": "http://rpi.thinlinx.com/NOOBS/TLX_RPi/TLXOS.png",
            "marketing_info": "http://rpi.thinlinx.com/NOOBS/TLX_RPi/marketing.tar",
            "nominal_size": 2064,
            "os_info": "http://rpi.thinlinx.com/NOOBS/TLX_RPi/os.json",
            "os_name": "TLXOS ",
            "partition_setup": "http://rpi.thinlinx.com/NOOBS/TLX_RPi/partition_setup.sh",
            "partitions_info": "http://rpi.thinlinx.com/NOOBS/TLX_RPi/partitions.json",
            "release_date": "2017-04-22",
            "supported_models": [
                "Pi 2",
                "Pi 3"
            ],
            "tarballs": [
                "http://rpi.thinlinx.com/NOOBS/TLX_RPi/boot.tar.xz",
                "http://rpi.thinlinx.com/NOOBS/TLX_RPi/tfm.tar.xz",
                "http://rpi.thinlinx.com/NOOBS/TLX_RPi/root.tar.xz"
            ]
        }
```
?

procount, 2017-04-26:
@matthuisman - there are no backdoors for promotions.
I guess some proxy is caching that file for me because I can still only see the version from 10-04-2017. I guess it will filter down eventually. @lurch - What date is the file you can see with this addition?

lurch, 2017-04-26:
'wget -S' says `Last-Modified: Tue, 25 Apr 2017 13:39:22 GMT`

XECDesign, 2017-04-26:
There is a problem with the mirrors, so the update to os_list_v3.json was only partially updated. No backdoors.

procount, 2017-04-26:
My mirror has finally updated and I can see it now.

matthuisman, 2017-04-26:
Same here.
Odd, my PC and Pi must of been using a different mirror (or it was Chrome caching it)

Seems odd that Raspberry Pi would have a non-free OS listed though....
Seems against the whole sort of non-profit style of things.

lurch, 2017-04-27:
> Seems odd that Raspberry Pi would have a non-free OS listed....?

Have you never noticed Windows IOT before? ;-)

matthuisman, 2017-04-27:
I have but that's free isn't it (just not open-source and requires Windows 10 on a PC?)

procount, 2017-04-28:
@XECDesign 
> The initial boot partition (NOOBS, PINN or whatever else), needs to have an up to date bootcode.bin and the boot partition you're booting into needs to have an up to date start.elf.

Is that the _only_ requirement in order to fix a non-autobooting OS - i.e. to have an appropriate start.elf to match the newer bootcode.bin in NOOBS? 
Or....
1) fixup.dat files should be updated to match start.elf?
2) dtbs have to updated to match start.elf?
3) Kernel has to be updated to match dtbs?

So what is the minimum that needs to be updated?
Or is it better to load the OS and perform a
`sudo apt-get update; sudo apt-get install --reinstall raspberrypi-bootloader`
But I prefer not to do that because it generally takes a lot longer and I may have to take care to preserve any run-once type startup tasks in the OS.

XECDesign, 2017-04-28:
fixup*.dat files always need to match start*.elf files.

Dtb files go with the kernel, so you shouldn't need to update either of those. 

Sometimes the kernel depends on a firmware version, but that wouldn't be the case here. 

When autoboot.txt is used, only bootcode.bin (in some cases, also parts of config.txt) is read off the NOOBS partition.

procount, 2017-04-28:
I did that on UbuntuMate and it seemed to fix it, but I went a stage too far and unnecessarily updated the DTBs as well. Although the bootcode.bin on the OS isn't loaded, I also updated that to be consistent with start*.elf (just in case). Thanks.

noge075, 2017-05-04:
Hi guys!

Is it possible to do any temporary solution for this problem?

procount, 2017-05-04:
If you are using NOOBS, I understand you will need to wait til v2.5.
PINN v2.4.1 already includes the necessary fix. But you need to modify the OS as well to be compatible.
I am currently modifying a lot of OS distros to be compatible with Autoboot ahead of their next releases. Most are easy, some don't work and you'll have to wait til the distros next update. Which OSes did you want to use?

noge075, 2017-05-04:
I just bought a RPi 3B v1.2 and using the given NOOBS v2.3. I installed the recommended Raspbian.

procount, 2017-05-04:
NOOBS v2.4 is already released, so you might want to upgrade to that. 
Raspbian already has the latest firmware so it **_might_** work in NOOBS 2.4 with autoboot already (I can't remember), but most of the other OSes will freeze with autoboot. Raspbian will definitely work with PINN 2.4.1

XECDesign, 2017-05-04:
The fix on the NOOBS end is just to grab a new bootcode.bin

[49 more replies]

commit eb9a7777 (2018-01-30): Bump to kernel 4.14

Update firmware
Fixes #390

Enable USB_RTL8152
closed 2018-09-24 by XECDesign