raspberrypi/noobs #393: fake-hwclock in install or rescue mode

jccolson, 2017-03-20:
I use a raspberry with a RTC and without ntpserver.
The date is always 1970-01-01 :-(
If a file "/etc/fake-hwclock.data" is created when a new version of Noobs are made,
/etc/init.d/S00hwclock added
and /init patched in recovery.rfs
Then the date can be automatically set to the building date of Noobs version.
It's now just for fun, but I hope to have i2c-tools in the future and do more with real hardware clock ;-)

[Uploading hwclock.zip…]()

jccolson, 2017-03-20:
/etc/init.d/S00hwclock
#!/bin/sh

# Fake hwclock
COMMAND=$1
FILE='/etc/fake-hwclock.data'
case $COMMAND in
    start)
        if [ -f "$FILE" ]; then
            date -u -s "$(cat $FILE)"
        fi
        ;;
    stop)
        # rootfs mounted readonly
        date -u '+%Y-%m-%d %H:%M:%S' > $FILE 2>/dev/null
        ;;
esac

jccolson, 2017-03-20:
init.patch
5a6,8
> # Enable hwclock
> /etc/init.d/S00hwclock start > /dev/null
>

jccolson, 2017-03-20:
With the zip file:
[hwclock.zip](https://github.com/raspberrypi/noobs/files/855528/hwclock.zip)

XECDesign, 2017-03-21:
Why? Is this to support HTTPS for custom os_list URLs?

jccolson, 2017-04-03:
It's for testing the availability of the right time in every stage of the raspbian's install.
I have added dtoverlay=i2c-rtc,ds1307 at the end of config.txt in boot.tar.xz for the raspbian and patched a script in root.tar.xz.
With this (plus setting the date before launching the install), all dates of partitions and files (logs, ...) are corrects.

XECDesign, 2017-11-03:
I don't think it matters what time NOOBS thinks it is when the network is not available. Happy to reopen is there is a good reason that I am unaware of.

closed 2017-11-03 by XECDesign