raspberrypi/noobs #394: Questions on latest commits

procount, 2017-03-29:
@maxnet  - I'm liking the look of the latest commits, but I have a few questions.
1. Is there an example of an os_list.json with the new "supports_xx" features in?
2. How will the user select where an OS root goes? (supported, but UI not implemented yet, or did I miss it?)
3. Is there an example of a new partition_setup.sh script? or will they be the same?
4. Why is fixate introduced?
Thanks.

maxnet, 2017-03-29:
>Is there an example of an os_list.json with the new "supports_xx" features in?

If your custom operating system only supports running from SD card:

```
supports_usb_root: false,
supports_usb_boot: false
```

If your custom operating system does support having it's root partition on USB disk, but does needs its boot partition on SD card:

```
supports_usb_boot: false
```

Only needed for very rare corner cases. 
E.g. if you have a need to keep using an old firmware version that does not support USB boot, or if your operating system needs to load extra modules from the boot partition on SD card before it is able to speak to USB.

If not specified both options default to true for Linux distributions. false for Windows/RiscOS.

>How will the user select where an OS root goes? (supported, but UI not implemented yet, or did I miss it?)

A dropdown "Destination drive" will appear in the main window near the available disk space indicator, if you have more than one drive.

>Is there an example of a new partition_setup.sh script? or will they be the same?

Same.
Your distribution should support PARTUUID as root= parameter though.

>Why is fixate introduced?

Normally not used.
If enabled end-users cannot change the operating system selection. Only reinstall the ones already selected.

procount, 2017-03-29:
> A dropdown "Destination drive" will appear in the main window near the available disk space indicator, if you have more than one drive.

Yes, I've seen that, but it only gives one destination.
What if the json file indicates boot & root are BOTH supported on USB, but the user wants the boot on SDcard and root on USB? How can he choose that?

Will it take account of the user having something other than a Pi3, in which case, the boot MUST be on SDcard because USB booting is not supported (even if json files says it is possible)?

> Your distribution should support PARTUUID as root= parameter though.

Yes, that was what I was worried about....

Also, if booting off USB with root on USB, should any OS that is not present be greyed out or hidden (e.g. maybe the right USB stick is not present)?

maxnet, 2017-03-29:
>Yes, I've seen that, but it only gives one destination.

What you select is the root drive.
Boot drive is the drive NOOBS booted from.

procount, 2017-03-29:
Ok. That'll work.
My last question was incorrect. Use case was booting off SDCARD with root on usb but wrong usb fitted so root not present.

maxnet, 2017-03-29:
>My last question was incorrect. Use case was booting off SDCARD with root on usb but wrong usb fitted 
>so root not present.

Then the OS being booted will simply wait for the user to plug the (right) drive in.
Probably indefinitely if ``rootwait`` is in cmdline.txt...

NOOBS itself doesn't wait for the drive, so wouldn't notice.
All information needed to display the menu is on the SD card, when booted from SD.
And waiting for a (potentially slow) drive to spin up, doesn't help with user experience.

procount, 2017-03-29:
Linux is cleverer than I thought!

closed 2017-03-29 by procount