raspberrypi/noobs #395: Append some extra "config" at the end of file config.txt in the installation stage

jccolson, 2017-04-03:
Example
- Create a file config.txt, for activating a specific RTC, in the Noobs's SD card witch contain :
dtoverlay=i2c-rtc,ds1307
- cat this file at the end of the config.txt extracted from boot.tar.xz before rebooting to the newly installed system.

Actually, I have to download Raspbian, patching config.txt in boot.tar.xz and put all stuff onto the directory /os/Raspbian of the SD card.

XECDesign, 2017-04-03:
NOOBS includes a config.txt editor. There are no plans to add any custom automated patching of config.txt files.

procount, 2017-04-03:
This sounds like you want to customise the OS (e.g. Raspbian) that is installed by NOOBS, but want to avoid creating a custom os?
Have you looked at my noobsconfig project which hooks into NOOBS to do just what you want?
https://github.com/procount/noobsconfig

closed 2017-11-03 by XECDesign