raspberrypi/noobs #396: OSMC broken

procount, 2017-04-09:
I know this isn't a NOOBS issue as the installation is now hosted by osmc themselves, but their partition_setup.sh script no longer fixes up the cmdline.txt and etc/fstab files. So if anyone has any contacts, perhaps they could pass this along...?

XECDesign, 2017-04-09:
Thanks, I'll pass it on.

XECDesign, 2017-04-09:
Are you sure there's a problem? It looks okay to me.

samnazarko, 2017-04-09:
Have you tested this with NOOBS? 

I've had a report of this on our forum [1], but it seems to be using 'PINN' which seems to be a modified version of NOOBS. Some changes were made to use labels for disks [2] instead of device nodes. I pushed this live recently, but haven't had reports of failed installations via NOOBS yet.

[1] https://discourse.osmc.tv/t/osmc-is-broken-if-installed-form-noobs/35992/3

[2] https://github.com/osmc/osmc/commit/04db2b30833b1a02d3486e6a6f8f877170222dbe

procount, 2017-04-09:
Yes, that user posted initially concerning PINN. So I tried it on PINN and it failed. So then I tried it on NOOBS v2.3 and it failed in the same way, hence I posted here. I have since tried it with my latest PINN that includes the NOOBS mods for disks and it still failed. I haven't tried the latest unreleased NOOBS with disks as I don't have it built at the moment.
In fact after installing, cmdline.txt and fstab seem to be non-existant.... very odd.

procount, 2017-04-09:
I'm trying the OSMC_PI2 version in case that makes a difference....

procount, 2017-04-09:
The environment for running the partition_setup script had the following (from debug): partuuid1=PARTUUID=00053966-06, partuiid2=PARTUUID=00053966-07, id1=LABEL=boot-rbp2, id2=LABEL=root-rbp2, part1=/dev/mmcblk0p6, part2=/dev/mmcblk0p7

So will lines like: 
```mount $id2 /tmp/mount ```
actually work? I think it would expand to:
```mount LABEL=root=rb2 /tmp/mount```

maxnet, 2017-04-09:
>mount $id1 /tmp/mount

In partition_setup.sh please use $part1 instead of $id1 when mounting, for compatibility with older versions of NOOBS.
While the $id variable has always been available, the ability to mount partitions by name has not been.

We were using busybox's built-in mount command previously, and that does not support mounting by label.
Should work with new version as we are using util-linux's mount there.

samnazarko, 2017-04-09:
Thanks for the replies.

>The environment for running the partition_setup script had the following (from debug): 

That looks correct. 

We've always used $part1 in the past, however we were told to start using $id1 for NOOBS' upcoming MSD support.

>We were using busybox's built-in mount command previously, and that does not support mounting by label.

It should do. OSMC's initramfs also uses BusyBox mount, and we are able to mount partitions by label. It does need to be explicitly enabled in the BusyBox config however if memory serves.

@procount You say that editing cmdline.txt and fstab is sufficient to boot post-installation. If this is the case then the busybox mount using labels is succeeding, but something's not working as expected with the sed rules. 

Do you want us to check a file, such as ```/etc/os-release``` to see if the NOOBS version we're running is capable of LABEL based mounting and fall back accordingly?

samnazarko, 2017-04-09:
If I'm reading correctly, you just want us to use $part variables for mounting.

procount, 2017-04-09:
@samnazarko - That's what the user mmhorda said. IIRC, he took the SD card out and made the modifications on his PC.
After NOOBS had installed OSMC, I found that the cmdline.txt and fstab files were actually missing. The sed rules looked fine to me.

@maxnet - I thought it was the new partuuid env vars that were added for msd, not the id ones? Shouldn't they be used somehow? Is there an example for their use?

samnazarko, 2017-04-09:
Okay -- for now I'm reverting ```partition_setup.sh``` to what we had earlier so that installs go through without issue now.

I'll keep an eye on this

XECDesign, 2017-04-09:
@procount either id or partuuid should work. id already existed, but hasn't been used by anyone. Since OSMC does not support PARTUUID, using LABEL(id) is the alternative. I will try it out in the morning and see whether there is a problem.

maxnet, 2017-04-09:
>If I'm reading correctly, you just want us to use $part variables for mounting.

Yes
$part1/$part2 for mounting
But do put $id1/$id2 in fstab/cmdline.txt

maxnet, 2017-04-09:
>It should do. OSMC's initramfs also uses BusyBox mount, and we are able to mount partitions by label. It 
>does need to be explicitly enabled in the BusyBox config however if memory serves.

If you enable it in config Busybox mount's can indeed handle ID/LABEL, and that should indeed be enough for you.
(It is not enough for NOOBS itself, because we needed PARTUUID support as well. So for us enabling util-linux's mount was preferred to patching busybox's configuration).

samnazarko, 2017-04-09:
Does OSMC need to implement PARTUUID support?

We can read the symlink to the real device from /dev/disk/by-partuuid, but I don't know whether $id1/$id2 will ever give us a PARTUUID or whether we need to worry about it.

maxnet, 2017-04-09:
>Does OSMC need to implement PARTUUID support?

No.
You are welcome to use ID instead.

Only Linux distributions that do not have a helper initramfs such as Raspbian *must* use PARTUUID.
That is because the Linux kernel does not understand ID/LABEL

samnazarko, 2017-04-09:
Okay -- I figured as much. 

I've now amended ```partition_setup.sh``` so that we still use the device nodes for mounting in NOOBS. I'm also not sure why @procount says fstab and cmdline weren't populated correctly, but I've added a fallback so if there is no $id1 or $id2, they'll at least get $part1 and $part2. 

So existing behaviour should be preserved when necessary. 

https://github.com/osmc/osmc/commit/e47836b8fa158214f5cbcbffb49b95fab102e192

maxnet, 2017-04-09:
>I've now amended partition_setup.sh so that we still use the device nodes for mounting in NOOBS. I'm 
>also not sure why @procount says fstab and cmdline weren't populated correctly, but I've added a 
>fallback so if there is no $id1 or $id2, they'll at least get $part1 and $part2.

$id1 / $id2 were always populated.
Just do not assume you can mount them.

samnazarko, 2017-04-09:
>$id1 / $id2 were always populated.

Makes sense. I thought cmdline and fstab were not being populated correctly, but looking above it seems @procount reports they weren't being created. That would happen if the mount failed; although I wouldn't expect a root filesystem to be properly present if that was the case either.

I've pushed it live now, and I'll ask the user in the thread to retry. 

Thank you @maxnet for the pointers

Sam

procount, 2017-04-10:
That's fixed it. Thanks.

closed 2017-04-10 by procount

samnazarko, 2017-04-10:
Good to hear. Thanks for testing.