raspberrypi/noobs #397: Lite v2.3 doesn't check for wifi

roryhr, 2017-04-10:
I have a Pi 3B and NOOBS Lite v2.3 but can't install anything. The error message says "Wired network access is required..." and once I click "Ok" the "wifi networks" button is greyed out.

XECDesign, 2017-04-13:
Does wifi work if you run Raspbian directly?

roryhr, 2017-04-13:
No. I suspect I have a hardware problem. It'd be nice if there was a test suite I could run to confirm.

XECDesign, 2017-04-13:
Could you post the output of dmesg and ifconfig?

daruvurisai, 2017-04-13:
I think There is a problem with hardware can you post the output of the following

roryhr, 2017-04-14:
Uhh. I guess it's working now? I didn't change anything. 

file:///home/pi/Desktop/ifconfig_dmesg_output.txt

roryhr, 2017-04-14:
It's working now and I'm not sure I changed anything since I last tried and it failed.

[ifconfig_dmesg_output.txt](https://github.com/raspberrypi/noobs/files/921514/ifconfig_dmesg_output.txt)

XECDesign, 2017-04-14:
Alright. Since this is not a NOOBS problem. Please post on the forum if you see it again.

closed 2017-04-14 by XECDesign

LickTheCheese, 2018-03-31:
i have the same problem, but when i run Raspian directly WiFi works.