raspberrypi/noobs #398: NOOBS 2.4 (minor)

procount, 2017-04-11:
You forgot to tag your commit as v2.4 before building NOOBS, so the BUILD-DATA file is probably not what you expected...

Build-date: 2017-03-17
NOOBS Version: **v2.1-28-g143b0fd**

XECDesign, 2017-04-11:
Yup, it's a minor annoyance. I would need to add annotated tags to releases before they're built. That means that if I need to build and test NOOBS and it turns out to be ready for release, I'd have to go back and rebuild it again, or cheat and manually modify the file.

It's a case where something that's meant to be nice and automated actually just creates more work that's easy to forget about. It should probably be the version and a simple short rev-parse instead.

procount, 2017-04-11:
Tell me about it! 
I have to be vigilant about it because I use it for the self-update, so mine has to be right. The number of times I find out I forgot it just before release.....

XECDesign, 2017-04-11:
Would a simple v$whatever_config.h_says (for example v2.4) instead cause you any problems?

XECDesign, 2017-04-11:
Actually, the `--short` doesn't contribute anything. I'll leave the full hash.

procount, 2017-04-11:
No, that should be fine. In my view, the config.h version, the tag annotation in git and the NOOBS version in BUILD-DATA should all be the same, so if they can come from a single source and be automated, all the better.

closed 2017-04-11 by XECDesign

lurch, 2017-04-11:
> I'd have to go back and rebuild it again

That's the approach that I used to use, but I guess the new approach works just as well :-)

procount, 2017-04-12:
So maybe running something like this (from /output) would be useful...

    #!/bin/sh 
    tag=$(sed -n 's|.*VERSION_NUMBER.*\"\(.*\)\"|v\1|p' ../recovery/config.h)    
    git tag $tag -m'$tag'

Or replace the annotation with some description or other.

commit 76539000 (2017-04-11): BUILDME.sh: get NOOBS version from config.h

Fixes #398