raspberrypi/noobs #402: keyboard and mouse combo don´t recognize in os selection menu

jor2404, 2017-04-18:
Hi, I have installed recalbox and raspbian using noobs.
During the installation I could use my keyboard and mouse combo but now that it is installed when I start the system in the selection os menu I can´t select the os that I want because the keyboard and mouse don´t work. But once in Raspbian they work perfectly.

XECDesign, 2017-04-19:
Not sure how that can happen without something outside of NOOBS going wrong. It works for me, so it's a bit difficult to investigate.

jor2404, 2017-04-19:
What could I do to discover what is going wrong?

XECDesign, 2017-04-21:
Does the mouse move? Does pressing shift take you to recovery? Does pressing ctrl-alt-f2 take you to a different console or is all input ignored entirely? Does it consistently work every time during installation and then fail when the boot selection screen comes up?

jor2404, 2017-04-21:
I have made all that tests but I didn´t get a response. I have connected to the raspberry two generic wireless pads that works perfectly in reclabox I disconnected them to check if they are relationed with my problem and after that if I press shift and then I can use the mouse and select the os to start. So, now that I know that something with my wireless pads blocks my keyboard and mouse in noobs Is there something that I can do to fix it?

XECDesign, 2017-04-25:
Sorry about the late reply.

Do you ever see a lightning bolt in the top right of the screen? I wonder if they wireless pads are drawing a lot of current and causing some usb devices to brown out.

lurch, 2017-04-26:
@XECDesign It's been a long time since I've looked at NOOBS, but is it possible that the USB HID handling isn't quite up to scratch, and something's getting confused by there being multiple input devices?

jor2404, 2017-04-26:
Is there anyway to make only vissible for Noobs the keyboard and mouse combo to avoid the confussion with other  connected usb devices?

XECDesign, 2017-04-26:
@lurch I've just tried attaching two usb keyboards and one wireless one and they all work without any problems. Input handling in linux is fairly straight forward. Another possibility might be that it's a problem with that specific device, but without having the problem in front of me, it's all guess-work.

jor2404, 2017-04-26:
The gamepads wich I think are the origin of this problem are generic ones. Could I made any test useful for you?

XECDesign, 2017-04-28:
@jor2404 I would boot noobs, press ctrl-alt-f2, login with root:raspberry, plug in the gamepads, verify that the keyboard has stopped working, unplug the gamepad and then run dmesg to see what's going on.

jor2404, 2017-05-06:
@XECDesign I made that and all seemed ok. Now I noticed that if I only have one of that generic pads the keyboard works but if I plugged both the keyboard doesn´t work at the os window selection

XECDesign, 2018-01-30:
A fix PR would be welcome, but closing it as a wontfix for now.

closed 2018-01-30 by XECDesign