raspberrypi/noobs #405: Win10 IoT

procount, 2017-05-12:
Regarding this topic on the forum ->
https://www.raspberrypi.org/forums/viewtopic.php?f=105&t=182639
Is Win10 IoT still supported in NOOBS, or has it been altered to the point that it is no longer viable with NOOBS?
I subsequently tried to install it and the installation came up with some errors about a couple of partitions.
It is still showing up in os_list_v3.json (4 May 2017), so if it is deprecated from NOOBS, maybe there is another mirroring problem?

XECDesign, 2017-05-12:
I've just installed it and it's fine. What model pi did you use?

procount, 2017-05-12:
I used a 3B, admittedly with PINN but that shouldn't make a difference. Odd that others had problems and reported that it was no longer supported. I'll have to try it again with pinn and noobs just in case.

procount, 2017-05-12:
I installed Win10IoT again with PINN and investigating the error messages more closely they were benign warning messages from PINN about post configuration. I chose the RTM retail version. It seemed to install ok. But when I booted into it, Win10 kept rebooting back into PINN. I'll try NOOBS next to see if there is any difference.

procount, 2017-05-12:
I installed Win10IoT with NOOBS-lite 2.4. On 1st boot, it checked drive U: then rebooted.
Then on the next boot, after the Windows logo screen, I get "The PC ran into a problem and needs to restart, We will restart it for you"  with a Bad system config info. Then it reboots and repeats. I give up. I don't use Win10IoT anyway.

XECDesign, 2017-05-12:
Did you install just Win10IoT by itself or with something else? Strange, since i used a pi 3B as well and it booted up properly. I'll see if I can ping MS about it.

procount, 2017-05-12:
With pinn I installed raspbian & win10. With noobs I just installed win10.

procount, 2017-05-13:
I had a thought and had another go, but this time I used PINN to install Win10IoT on a RPi2B with RPF touchscreen using 2.5A PSU, instead of running RPi3B off my Motorola Lapdock.  Apart from my annoying PINN messages, it installed fine and booted fine. I think my previous boot loops must have been due to a lack of power issue, so it looks OK to me.

XECDesign, 2017-05-13:
Thanks for double checking.

closed 2017-05-14 by procount