raspberrypi/noobs #407: Add to the repository translation_en.ts template language file for new translation

bovirus, 2017-06-02:
@XECDesign 

To help the translator to develop new language I believe that could be useful to add to the repository a

translation_en.ts

template language file (without strings translated).

I created this file converting one .TS language file in .PO with lconvert command.
The using Poedit a created a template (removing all tramslations).
Then I converted the po template file in .TS format using lconvert command.

lurch, 2017-06-02:
Does https://github.com/raspberrypi/noobs#addingupdating-translations not already cover this process?

bovirus, 2017-06-02:
@lurch 

Many user has no significant knowledge to do all steps.

To encourage the help of many translators as much as possible, I believe the simple step could be create a small packahage that contain

QTLinguist standalone
translation_en.ts (empty)

Anyone can use QTLinguist and pos here the new file.
You or other user with a big knoledge can do all other steps

XECDesign, 2017-11-03:
I think the existing instructions are sufficient.

closed 2017-11-03 by XECDesign