raspberrypi/noobs #415: Add Screenly OSE image

vpetersson, 2017-06-16:
Hey guys,

I got a working [Screenly OSE](https://github.com/Screenly/screenly-ose) image for Noobs. Just need to put some final touches on it (add graphics etc). However, I would like to see what the process is for adding that image upstream. How can I contribute this image? I'm happy to put it on a CDN myself.

vpetersson, 2017-06-16:
After a closer look, it seems like this is all managed in http://downloads.raspberrypi.org/os_list_v3.json

Who's in charge of that?

XECDesign, 2017-06-19:
Hi Viktor, we're happy to add Screenly to NOOBS. We'll just need the appropriate snippet for os_list_v3.json.

vpetersson, 2017-06-19:
@XECDesign Excellent. Will post it here this week.

vpetersson, 2017-06-20:
Here's a draft of the JSON payload. I think that is correct, but it is hard to test it without a MiTM :)

```
        {
            "description": "Screenly Open Source Edition (OSE) - The most popular digital signage platform for the Raspberry Pi",
            "icon": "https://releases.screenlyapp.com/NOOBS/Screenly_OSE.png",
            "marketing_info": "http://releases.screenlyapp.com/NOOBS/marketing.tar",
            "nominal_size": 4002,
            "os_info": "https://releases.screenlyapp.com/NOOBS/os.json",
            "os_name": "Screenly OSE ",
            "partition_setup": "https://releases.screenlyapp.com/NOOBS/partition_setup.sh",
            "partitions_info": "https://releases.screenlyapp.com/NOOBS/partitions.json",
            "release_date": "2016-12-01",
            "supported_models": [
               "Pi Model",
               "Pi 2",
               "Pi Zero",
               "Pi Compute Module",
               "Pi 3"
            ],
            "tarballs": [
                "https://releases.screenlyapp.com/NOOBS/boot.tar.xz",
                "https://releases.screenlyapp.com/NOOBS/root.tar.xz"
            ]
        }
```

Let me know if you need anything from me.

Thanks.

procount, 2017-06-20:
You can always create your own os_list_v3.json that includes the above contents and put it in the NOOBS recovery partition. NOOBS will load such a file for testing purposes.

vpetersson, 2017-06-20:
Thanks, @procount. Tested and confirmed working.

XECDesign, 2017-06-20:
Should be added tomorrow. Thanks.

vpetersson, 2017-06-20:
👍

vpetersson, 2017-06-21:
@XECDesign could you ping me here when it's live?

XECDesign, 2017-06-21:
@vpetersson ping

Should be live now. I'd recommend reducing the description length so that it fits in the window. Just "A digital signage platform" would be good, since the name is already listed and the users just need to know what it is.

vpetersson, 2017-06-21:
@XECDesign Awesome! Good comment about the description. How many characters can fit?

XECDesign, 2017-06-21:
Going above 70 seems starts to go over the ethernet icon and then off the box.

vpetersson, 2017-06-21:
Thanks. Good call.

Let's do this instead:

```
"description": "The most popular digital signage platform for the Raspberry Pi",
 "os_name": "Screenly Open Source Edition (OSE)",
```

XECDesign, 2017-06-21:
Sounds good. Will update it in a bit. Just checking if usb root, sd boot works. I've noticed usb boot doesn't, so os_list will need a line that specifies that.

vpetersson, 2017-06-21:
>  I've noticed usb boot doesn't

You mean installing the Screenly onto a USB drive? That should work as far as I know. If not, I will have to take a look.

XECDesign, 2017-06-21:
NOOBS supports three options
1) Everything on SD
2) Boot on SD and root on USB
3) Everything on USB
1 and 2 will just work in most cases, but 3 requires that partitions_setup takes it into account and that the OS supports either PARTUUID= or LABEL=. When I tried, I just got a black sreen.

vpetersson, 2017-06-21:
Ah, I see. Ok, let me take a look at that later.

XECDesign, 2017-06-21:
Heads up that 2 doesn't work either. the kernel boots, but after the plymouth splash, it drops to emergency mode.

vpetersson, 2017-06-21:
Ok, I will create a new disk image when we release the next version of OSE. It's coming close now as we are at the end of the sprint. Let's stick to 1 for now and then I'll QA 2 and 3.

Thanks for the help.

procount, 2017-06-21:
> Let's do this instead:
> 
> "description": "The most popular digital signage platform for the Raspberry Pi",
>  "os_name": "Screenly Open Source Edition (OSE)",

@vpetersson : The os_name looks overly long to me. Remember that it is often used as the base name for the OS folder, icon name etc. (by default if not overridden). I would suggest keeping your original "Screenly (OSE)".

vpetersson, 2017-06-21:
Sure. Let's just call it "Screenly OSE".

closed 2017-06-21 by vpetersson