raspberrypi/noobs #420: gentoo-on-rpi3-64bit add to noobs.

necrose99, 2017-06-23:
https://github.com/sakaki-/gentoo-on-rpi3-64bit

rather usable Image 

only Caveat is partitioning control. 
/boot / (root) **current state**...  (fat32 /boot)  root @ ext3/4 

/efi /boot / (rootfs)    ,   /boot/efi , boot 550 meg  , root growable.  wishlist...  
(and or use desired parts.. ) 

some of the distro kernels have RPI3 firmware in them , so testing them is kinda a pain 

hence it would be nice to have a script to migrate to thus. or have noobs do a more flexible layout.

XECDesign, 2017-06-23:
We tend to stick with files packaged and maintained officially by the project.

Although I started out as a Gentoo user and appreciate the educational value of setting up and maintaining a Gentoo install, I am not quite convinced it's a practical OS for the pi.

I do think it's likely to be picked up by PINN, if it hasn't been already though.

procount, 2017-06-23:
I'll have a look...

necrose99, 2017-06-23:
https://wiki.gentoo.org/wiki/Raspberry_Pi_3_64_bit_Install @sakaki-  has been a well known contributer to Gentoo , and as far as the main Gentoo wiki , has been pretty much maintaning it. 

https://github.com/sakaki-/gentoo-on-rpi3-64bit/releases/download/v1.0.2/genpi64.img.xz 
also has a limited number of bins.  on offer for more basic things. 

Etcher.io , z7ip rufus work on windows 10.  granted a pi3 compatible device odroid/Banana pi would yield better building. 2 gigs 
 but out of the box for light apps vlc , snes9x etc,  unfortunately there's not a RPI3-EX for more ram,  

so far I've been building from pentoo and docker etc. for a 64gig card. , granted most of it will get migrated over.  to a full arm64 server but for and Inexpensive test-case the pri3 has done well for building up a repo pile.  
but as far as an official ISO for arm64 Gentoo is lagging.   and far as gentoo installs vs sabayon (gentoo) 
gentoo is a more DIY Linux from scratch...   easier with ISO's .. much like Archlinux..

procount, 2017-06-23:
I had a quick read of the first linked page. Although it is ARM64, it seems not to use U-BOOT like many others, so that's a good start. Provided I can move and edit the partition references it should be do-able. I've never used Gentoo though, but if it is similar to debian/raspbian I should be able to find my way around.

necrose99, 2017-06-23:
Sabayon Linux (Gentoo based )  maintains  more conventional 32bit Images. 
https://www.sabayon.org/article/special-release-sabayon-1606-arm/  ,  equo up equo i app-admin/??? 
muck like rpm ,  but also can use reg gentoo tools if you need to build.   a more intermediate-to advanced distro. 

however , I've pushed packages and reports @ devs. , but I expect not as much so soon.  
by namilly force building some of the keystone packages...  (basic sabayon Infra.. ) 

however many are familiar with Gentoo or Pentoo (gentoo Pentesting) (takes a bit more as pentoo hasn't an arm-7/64 repo at present , granted Kahli-pi3 is easier , but 9999 packages ensure git-clone ./conf.. make etc.  -*Freshly-Squeezed*-  )  , the pi3 install from scratch is a bar to using them..  being a more advanced distro...

closed 2017-06-27 by XECDesign