raspberrypi/noobs #422: Raspbian /var/lib/dpkg/available empty after installation

newdwarf3, 2017-06-28:
After using NOOBS 2.4 to install Raspbian and Openelec using the versions included in the image, the package management data is partially broken: /var/lib/dpkg/available is empty.
So you can't use dpkg -p to get information about an installed package.

XECDesign, 2017-06-29:
I am not sure that this is accurate. I can see the file in the Raspbian archive that NOOBS extracts.

newdwarf3, 2017-06-29:
That may be true. But immediately after installation, at 1st boot into the **installed** Raspbian, the file exists but has file size 0 (= empty). Either it was already size 0 in the image, or it was broken during the installation process. I tried three fresh installs now, size is always zero.

XECDesign, 2017-06-30:
Thanks for double checking. I'm looking into it.
I've just downloaded NOOBS Lite, started an install of Raspbian and Raspbian Lite. While installing, I can see that the file is there and contains valid data. Before NOOBS rebooted, the file was still there. Booted into Raspbian - the file is there. Booted into Raspbian Lite - the file is there.

There is no possible mechanism that I can think of which would result in that specific file getting removed.

lurch, 2017-06-30:
I wonder if maybe the file is being updated while @newdwarf3 is checking it, which might result in it appearing to have a size of 0 bytes? (guess)

newdwarf3, 2017-06-30:
@lurch : I thought the same, but there is no activity on the file or system in general.
I did some more tests: I selected the network versions as @XECDesign did (instead the included images as I did before). Indeed, the resulting  /var/lib/dpkg/available are not of size zero after that. But they are still not complete: they contain information for only a fraction of the installed packages. If you check with dpkg -l what is installed, and then do some spot checks you will see:
"dpkg -p adduser" is fine: information is in the file and will be shown
"dpkg -p xauth" is installed but this throws an error message.
 
I can only guess that the initial install process that created the Raspbian master images for NOOBS is somehow weird.

maxnet, 2017-06-30:
Just to be sure, have you also tried a different SD card?

As certain write errors may go unreported.
Keep in mind that kernel caches writes to memory first, and will report success to the application, even if it hasn't started to actually commit the data to storage.

newdwarf3, 2017-06-30:
Yes, I tried two different cards. I also installed two OSes (Raspbian and Raspbian Lite) on one card in parallel, and it would be very unlikely that everything is fine on both partitions except these two files ;-)

maxnet, 2017-07-01:
>Yes, I tried two different cards.

And you also experienced the problem that the file was empty with both of them?
Not just that you believe the file was incomplete?

==

>I did some more tests: I selected the network versions of Raspbian and Raspbian Lite as 
>@XECDesign did (instead the included images as I did before). Indeed, the resulting 
>/var/lib/dpkg/available are not of size zero after that. But they are still not complete: they contain 
>information for only a fraction of the installed packages.  If you check with dpkg -l what is installed, and then do some spot checks you will see:
>
>"dpkg -p adduser" is fine: information is in the file and will be shown
>"dpkg -p xauth" is installed but this throws an error message because the info is missing.

My guess is that `xauth` was installed by `apt-get` which has it's own meta data, and may not bother updating `dpkg`'s available file.
While `adduser` was installed by `debootstrap`.
If so, that has little to do with NOOBS.

lurch, 2017-07-02:
@newdwarf3 If you're sure there's a problem, perhaps it would also be worth checking with 'Raspbian standalone' (i.e. non-NOOBS) and / or older versions of NOOBS, to see if different / earlier versions behaved differently?

closed 2017-07-03 by XECDesign

lurch, 2017-07-17:
@XECDesign Is this possibly related to https://github.com/raspberrypi/documentation/issues/693 ?

XECDesign, 2017-07-17:
As you know, NOOBS installs from archives, but that issues only affects images. The content of the archives is valid, so I don't think that's it.