raspberrypi/noobs #423: Please reconsider locale bug #421

lipmane, 2017-06-30:
Please reconsider the "wontfix" status of the locale bug I described in #421.  Consider the
following situation from the perspective of a not very well informed new user:

1. User selects "English (US)" with "us" keyboard, and installs Raspbian using NOOBS 2.4.1.
2. User changes default password and confirms using passwd or another non-echoing
    utility.
3. Shortly thereafter, user discovers that the keyboard is not functioning as expected.
    For example, Shift-3 produces a British pound symbol rather than a hash mark.
4. The confounded new user seeks help from a more experienced friend, who uses
    raspi-config to change the locale to en_US.UTF-8, as it should have been.
5. The new user soon attempts to use his password, and finds it is no longer accepted.
    He is certain that he had a hash mark (#) at the end of his password...

Several of the students in my class, which begins with a NOOBS installation, have
now had similar experiences.  I think this bug is serious enough to warrant some
sort of attention in the near future.

XECDesign, 2017-06-30:
It is definitely on the todo list, I just can't promise that it will be resolved in the next Raspbian release. It is still not obvious why it doesn't happen automatically, so there is some time that needs to be set aside from other things to investigate this properly.

lurch, 2017-06-30:
@lipmane From what I remember of the way that Raspbian and NOOBS interact, this would actually be a bug in Raspbian rather than a bug in NOOBS. I guess you could create an issue at https://github.com/RPi-Distro/pi-gen ?

XECDesign, 2017-07-10:
I think I see what the problem is. On one boot, I've seen the service time out, which meant that the locales weren't properly generated. Even when the service does finish, it happens after the desktop and/or getty start, so the new locale is not used. Porting the init.d script to systemd should provide some options to work around this, at the cost of a much longer first boot.

lipmane, 2017-07-15:
Thank you, I'll do that.

Everett

lipmane, 2017-07-15:
Thank you for the update.

XECDesign, 2017-07-15:
A bit of a further update, there were three separate issues which prevented this from working, which took a while to track down. The next release should have the fixes.

lipmane, 2017-07-15:
Great!  Thank you.

Everett

XECDesign, 2017-08-17:
Should be fixed with today's Raspbian release.

closed 2017-08-17 by XECDesign

ktjr, 2017-08-17:
When is the release of Latest Raspbian?
Is it possible to know what all are updates in the latest version?
Am so excited to know of it!!!

XECDesign, 2017-08-17:
https://www.raspberrypi.org/blog/raspbian-stretch/
downloads.raspberrypi.org/raspbian/release_notes.txt