raspberrypi/noobs #424: New Screenly OSE release (fixes USB boot etc)

vpetersson, 2017-07-04:
Hey guys,

This is a follow-up to #415. As correctly pointed out in that issue, there were issues with both installing to USB and booting from USB in the previous release.

I've already pushed the new disk images based on the [Night Train release](https://github.com/Screenly/screenly-ose/releases/tag/v0.10) and confirmed that it now works.

As a technical side-note, this build is based on the 2017-06-21 release of Raspbian Jessie Lite.

New features:

* Fixes installation to USB
* Fixes installation boot from USB
* Adds support for the Zero W

Diff for `os_list_v3.json` (i'm not sure if `release_date` should be bumped up with every release or not): 

```
{
[...]
     "release_date": "2017-07-04",
      "supports_usb_root": "true",
      "supports_usb_boot": "true",
[...]
}
```

XECDesign, 2017-07-05:
Thanks Viktor, we've updated the file accordingly.

We don't have to bump release_date every time, since the version in your os.json takes precedence.

vpetersson, 2017-07-05:
Got it. Thanks.

closed 2017-07-05 by vpetersson