raspberrypi/noobs #427: Fails to build with gcc version 6.3.0 20170516 (Debian 6.3.0-18)

voltagex, 2017-07-19:
```
/usr/bin/g++ -c  -DIN_GCC_FRONTEND -O2 -I/home/voltagex/src/noobs/buildroot/output/host/usr/include -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -Icp -I../../gcc -I../../gcc/cp -I../../gcc/../include -I../../gcc/../libcpp/include -I/home/voltagex/src/noobs/buildroot/output/host/usr/include -I/home/voltagex/src/noobs/buildroot/output/host/usr/include -I/home/voltagex/src/noobs/buildroot/output/host/usr/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace    ../../gcc/cp/init.c -o cp/init.o
In file included from ../../gcc/cp/except.c:1008:0:
cfns.gperf: In function ‘const char* libc_name_p(const char*, unsigned int)’:
cfns.gperf:101:1: error: ‘const char* libc_name_p(const char*, unsigned int)’ redeclared inline with ‘gnu_inline’ attribute
cfns.gperf:26:14: note: ‘const char* libc_name_p(const char*, unsigned int)’ previously declared here
cfns.gperf: At global scope:
cfns.gperf:26:14: warning: inline function ‘const char* libc_name_p(const char*, unsigned int)’ used but never defined
In file included from ./tm.h:26:0,
                 from ../../gcc/cp/init.c:26:
../../gcc/config/elfos.h:102:21: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
    fprintf ((FILE), "%s"HOST_WIDE_INT_PRINT_UNSIGNED"\n",\
                     ^
Makefile:1059: recipe for target 'cp/except.o' failed
make[3]: *** [cp/except.o] Error 1
../../gcc/config/elfos.h:170:24: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
       fprintf ((FILE), ","HOST_WIDE_INT_PRINT_UNSIGNED",%u\n",  \
                        ^
make[3]: *** Waiting for unfinished jobs....
In file included from ./tm.h:44:0,
                 from ../../gcc/cp/init.c:26:
../../gcc/defaults.h:126:24: warning: invalid suffix on literal; C++11 requiresa space between literal and string macro [-Wliteral-suffix]
       fprintf ((FILE), ","HOST_WIDE_INT_PRINT_UNSIGNED",%u\n",  \
                        ^
rm gcc.pod
make[3]: Leaving directory '/home/voltagex/src/noobs/buildroot/output/build/host-gcc-final-4.8.4/build/gcc'
Makefile:3909: recipe for target 'all-gcc' failed
make[2]: *** [all-gcc] Error 2
make[2]: Leaving directory '/home/voltagex/src/noobs/buildroot/output/build/host-gcc-final-4.8.4/build'
Makefile:857: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/voltagex/src/noobs/buildroot/output/build/host-gcc-final-4.8.4/build'
package/pkg-generic.mk:182: recipe for target '/home/voltagex/src/noobs/buildroot/output/build/host-gcc-final-4.8.4/.stamp_built' failed
make: *** [/home/voltagex/src/noobs/buildroot/output/build/host-gcc-final-4.8.4/.stamp_built] Error 2
voltagex@devbox:~/src/noobs$
```

voltagex, 2017-07-19:
My mistake, Debian testing is now on gcc 6.4.0.

voltagex, 2017-07-19:
Same issue with gcc 6.4.0.

https://bugs.debian.org/cgi-bin/bugreport.cgi?att=1;bug=831142;filename=gperf-gcc-6.patch;msg=10 is the required patch.

closed 2017-11-02 by XECDesign

XECDesign, 2017-11-02:
For me, it's failing later in the build process now. It might be time to do a complete refresh and update buildroot.

XECDesign, 2017-11-02:
Should be actually fixed now.

commit fc209357 (2017-11-02): Fix gcc >6.3.0 builds

Fixes #427

Automatically updated translations files