raspberrypi/noobs #429: missing file translation_ca.qrc

voltagex, 2017-07-31:
```/home/ubuntu/noobs/buildroot/output/host/usr/bin/rcc -name icons icons.qrc -o qrc_icons.cpp
RCC: Error in 'icons.qrc': Cannot find file 'translation_ca.qm'
Makefile:386: recipe for target 'qrc_icons.cpp' failed
make[1]: *** [qrc_icons.cpp] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/ubuntu/noobs/buildroot/output/build/recovery-1.0'
package/pkg-generic.mk:182: recipe for target '/home/ubuntu/noobs/buildroot/output/build/recovery-1.0/.stamp_built' failed
make: *** [/home/ubuntu/noobs/buildroot/output/build/recovery-1.0/.stamp_built] Error 2```

bovirus, 2017-07-31:
The translation file is in place.
Please delete the localfolder and try again to get the full structure of noobs.

XECDesign, 2017-08-07:
@voltagex can this be closed?

voltagex, 2017-08-07:
Sure, I haven't had a chance to try this again, I'll re-open if it reoccurs.

closed 2017-08-07 by XECDesign

Immortalin, 2017-08-23:
@XECDesign still getting the same error

XECDesign, 2017-08-24:
@Immortalin builds fine for me. Please provide steps to reproduce the problem starting from a clean copy of the repo.